MLB-LED-Scoreboard/mlb-led-scoreboard #311: Does this work with daisy chained LED panels?

rgrech35, 2021-02-04:
Hi I have two Adafruit 64x32 LED panels that I used for another project. Will this package work with daisy changed boards? I was hoping to get an answer on this before buying some of the additional parts I would need for this project.

Thank you.

ty-porter, 2021-02-04:
Yes, it will work: 

See the flags in the [driver library](https://github.com/hzeller/rpi-rgb-led-matrix/tree/6668ff8b55fcfe5ea36cf22aa48155c9ced177a9#panel-connection) as well as [this repo](https://github.com/hzeller/rpi-rgb-led-matrix/tree/6668ff8b55fcfe5ea36cf22aa48155c9ced177a9#panel-connection).

For your purposes the total board size would be 64x64 or 128x32, both of which have supported layouts.

rgrech35, 2021-02-04:
> See the command line flags in the driver library

Thanks so much!

closed 2021-02-05 by ajbowler