MLB-LED-Scoreboard/mlb-led-scoreboard #314: Standings broken when MLB API returns null

tracerrx, 2021-03-05:
Apparently if standings are enabled during spring training python script chokes... Needs to die more gracefully.

> ERROR (2:26:46): Could not render standings.  Falling back to off day.
> ERROR (2:26:46): StopIteration: ()

ty-porter, 2021-03-05:
Dumping a whole bunch of Slack info here to help debug:

https://statsapi.mlb.com/api/v1/standings?season=2021&leagueId=103,104&date=03/05/2021&division=all
Returns an empty `results` array, doesn't appear to be the root of the issue, this appears to be gracefully handled by the renderer (see below)

---

Relevant renderer code:

https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/44f6ebd00930ed1da99a3179fc5ea9e734d5e05a/renderers/main.py#L75-L83

`try/except` block failing to render standings logs debug message from this issue -- next callout is to `__render_offday()`, seems like a likely candidate for error.

---

Disabling offday rendering causes things to begin processing but no output to the LEDs:

[SLACK](https://mlb-led-scoreboard.slack.com/archives/C019A7T6J21/p1614620650024300)
```
INFO (11:42:56): MLB LED Scoreboard - v4.0.1 (64x32)
INFO (11:42:56): Custom '64x32.json' found. Merging with default reference layout.
DEBUG (11:42:56): Updating games for 2/26/2021
DEBUG (11:42:57): Weather should update!
DEBUG (11:42:57): API Key hasn't been flagged as bad yet
DEBUG (11:42:57): Weather: 48F; Wind: 10mph NE; Clear (Assets/weather/01d.png)
DEBUG (11:42:57): Headlines should update!
DEBUG (11:42:57): 3 feeds to update...
DEBUG (11:42:57): Feed URLs found...
DEBUG (11:42:57): Fetching https://www.mlb.com//feeds/news/rss.xml
DEBUG (11:42:58): Fetched feed 'MLB News' with 25 entries.
DEBUG (11:42:58): Fetching https://www.mlb.com/nationals/feeds/news/rss.xml
DEBUG (11:42:59): Fetched feed 'Nationals News' with 25 entries.
DEBUG (11:42:59): Fetching https://www.mlbtraderumors.com/washington-nationals/feed/atom
DEBUG (11:42:59): Fetched feed 'Washington Nationals  MLB Trade Rumors' with 15 entries.
```

twfarley, 2022-04-08:
No longer using mlbgame with the latest version.

closed 2022-04-08 by twfarley