MLB-LED-Scoreboard/mlb-led-scoreboard #315: Unable to access config.json

WashCaps33, 2021-03-16:
I have had no luck accessing the config.json file to make my customizations.

pjockey, 2021-03-17:
How are you trying to access it? I've always changed to /nhl-led-scoreboard/config and then used nano. 
cd
nano /nhl-led-scoreboard/config/config.json 
use ctrl+o, then enter to save the changes and keep nano open, or ctrl+x and y to save and exit.

ty-porter, 2021-03-17:
> I have had no luck accessing the config.json file to make my customizations.

I am afraid you'll need to be more specific than this for us to help you.

Are you saying there's an issue with the installer generating the configuration? Or are you just asking for general help on how to edit the configuration? 

If it's the latter you may want to check out the [Slack channel](https://mlb-led-scoreboard.herokuapp.com/), Github issues probably isn't the place to discuss this.

> How are you trying to access it? I've always changed to /nhl-led-scoreboard/config and then used nano.

Wrong scoreboard my friend, but good advice in general!

WashCaps33, 2021-03-17:
No I am trying to do it in the MLB board.  I assumed it was the same setup as the NHL version, but not so in this case. Have absolutely no idea how to get where to customize anything.

pjockey, 2021-03-17:
```
pi@raspberrypi:~/mlb-led-scoreboard $ ls
Assets               data       install.sh  main.py    renderers
config.json          debug.py   ledcolors   Makefile   startup.sh
config.json.example  debug.pyc  ledcoords   matrix     utils.py
CONTRIBUTING.md      img        LICENSE.md  README.md  utils.pyc
pi@raspberrypi:~/mlb-led-scoreboard $ 
```
nano config.json

That's it.

ty-porter, 2021-03-17:
> No I am trying to do it in the MLB board. I assumed it was the same setup as the NHL version, but not so in this case. Have absolutely no idea how to get where to customize anything.
> 
> I want to make changes to the config

@WashCaps33 

Again, unless you **specifically** tell us what you're doing, there is very little chance we'll be able to help. This project is not the same as the NHL Scoreboard and you should not assume the instructions for setting it up will apply here.

WashCaps33, 2021-03-17:
I wasn't trying to rustle any feathers so I apologize.  I'm new to this stuff and trying to understand.  Figured it's best to ask the pros.  Sorry about that.

pjockey, 2021-03-17:
@WashCaps33 the only thing is, this is the nhl-led-scoreboard program and the mlb-led-scoreboard has its own github set up. I think they have a #Slack channel/chat that can be used to ask questions. 

I didn't have a problem answering you here tho. There is not much to configure in the mlb config.json file.

WashCaps33, 2021-03-17:
I understand.  But I have installed the MLB version, completely independent of the NHL one.  Being new, I thought coding was relatively similar.  I'm just having a rough time getting the led to show anything but the date and time.  Just frustrated so I apologize

ohjk, 2021-03-17:
You need to come to slack if you need help like this

ty-porter, 2021-03-17:
> I understand. But I have installed the MLB version, completely independent of the NHL one. Being new, I thought coding was relatively similar. I'm just having a rough time getting the led to show anything but the date and time. Just frustrated so I apologize

@WashCaps33 Please visit our [Slack channel](https://mlb-led-scoreboard.herokuapp.com/), it is totally reasonable to have questions and we will be much more equipped to help you there. 

Github issues are for issues with the software itself, not questions about how to use the software.

@pjockey This **is** the mlb-led-scoreboard repo, this user is in the correct place for MLB scoreboard questions (although I would prefer this get taken to Slack based on the content of the issue).

closed 2022-04-08 by twfarley