MLB-LED-Scoreboard/mlb-led-scoreboard #316: LED flicker and glitching

MarkBeach6, 2021-03-23:
I have a 64x32 led board with a 5mm pitch and it works fine and I can get the program running, but it often flickers and glitches
[IMG-0343 - Copy.zip](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/6191597/IMG-0343.-.Copy.zip)

ty-porter, 2021-03-23:
Couple suggestions:

This is generally a hardware issue, so not something we can diagnose over the internet. I'd try re-seating any cables and ensuring there's no cold solder joints (if you've done any modifications). Long data lines can be an issue, so you should replace them with shorter ones if they're too long.

If you're certain the hardware works:

* Ensure you're using the right `--led-gpio-mapping` command line argument
  * https://github.com/hzeller/rpi-rgb-led-matrix/tree/6668ff8b55fcfe5ea36cf22aa48155c9ced177a9#changing-parameters-via-command-line-flags
  
* Try slowing down the GPIO speed
  * https://github.com/hzeller/rpi-rgb-led-matrix/tree/6668ff8b55fcfe5ea36cf22aa48155c9ced177a9#gpio-speed
  
You should check out our [Slack channel](https://mlb-led-scoreboard.herokuapp.com/) for further troubleshooting

MarkBeach6, 2021-03-24:
I changed the GPIO speed like you said and it seemed to work. Thanks!

thomasr05, 2021-03-26:
Seems I have the same issue. I'm using a Pi4, Bonnet, 32x32 matrix, 5v 4amp charger and still having no luck.

[Video of the flicker](https://drive.google.com/file/d/1KU58KrPQ6QGxX3uChed7LeGU8vio8X-6/view?usp=sharing)

### What I Have Tried
- I am always using **--led-gpio-mapping="adafruit-hat"** to start the board
- I have disabled the audio in the boot config.txt, so I could make the adjustments.
- Changed out my cables
- Tried setting the GPIO speed from 0-4, so far the default 1 seems to be the best. Despite ADA say 4 for a pi4, it looks completely jacked up.
- Tried dimming the lights nothing really happened to fix it.
- My refresh rate is 230hz... (not sure what the baseline would be, I think 250 decent that I saw)
- Tried adding dtoverlay=w1-gpio gpiopin=19 in the boot config.txt
- I have both my plugged into the pi and the bonnet plugged in (Before just the bonnet gave me low voltage warnings)

### What I haven't tried yet
1. Going headless, although I wouldn't think it would be an issue for a pi4
2. Doing a jumper - https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/driving-matrices

Any thoughts?

pjockey, 2021-03-26:
I would do the jumper. Takes couple of minutes to do.

thomasr05, 2021-03-29:
Did the jumper, the flickering drop a lot but was still there.

I follow the Troubleshooting section and I noticed it made a difference in turning off services. They made the comment you should install the light OS because of software conflicts.
https://github.com/hzeller/rpi-rgb-led-matrix

 I might try headless next.

thomasr05, 2021-04-01:
I figured out the flickering! Mainly because of me being an idiot. I didn't know how to use flags correctly. I was running the service twice rather than once causing all the issues. 

However, I now see red pixels in the same spots. Any thoughts?

https://drive.google.com/file/d/1bOCfjvIu3aokXYlxejkkCdQr6oq0Q0Rw/view?usp=sharing

`--led-gpio-mapping="adafruit-hat" --led-slowdown-gpio=4 --led-brightness=90`

pjockey, 2021-04-01:
Place a -pwm after the adafruit-hat.  Don't think you need the quotation marks around adafruit-hat either 

So like --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=4 --led-brightness=90

thomasr05, 2021-04-01:
Sadly I tried it and the board goes dark everytime I use  --led-gpio-mapping=adafruit-hat-pwm
I guess I can't even solder two simple pins correclty.... 
Do you think it's because the solder bleeding through the 18pin?
Front - https://ibb.co/jyvLHfy
Back - https://ibb.co/Y3XG0Lq

pjockey, 2021-04-01:
That looks ok

pjockey, 2021-04-01:
sudo python main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64

This is my command that works with no flicker or red lines.

twfarley, 2022-04-08:
Sounds like this was resolved. See https://github.com/hzeller/rpi-rgb-led-matrix for more tips on how to reduce flicker.

closed 2022-04-08 by twfarley