MLB-LED-Scoreboard/mlb-led-scoreboard #317: 64x128 board overlapping

I3lackshirts, 2021-03-26:
[image]x

see the attached picture. Basically what is happening is the first 16 rows are displaying both the first 16 rows and rows 17-32 on top of each other. Same thing is happening for the bottom half. I tried inputting a different led-panel-type and just was getting errors. Anyone run into this before and know what to try?

twfarley, 2021-03-26:
Yeah, I know exactly what's happening. Your HAT doesn't support 64 height. Depending on when you bought it you can modify it to support that much. Share a picture of your HAT and I'll show you what you need to do.

I3lackshirts, 2021-03-26:
[image]

I3lackshirts, 2021-03-26:
[image]

twfarley, 2021-03-26:
Well, I thought I knew but that was when I thought you were using the Adafruit HAT. You'll have to consult the manufacturer of that HAT to find out how to use the E pin for multiplexing.

ty-porter, 2021-03-27:
Documentation for that adapter is [here](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#choose-e-line-for-64x64-panels-with-132-multiplexing).

I3lackshirts, 2021-03-27:
That was the issue. I connected the E_BUF and P4 pins. Thanks everyone!

closed 2021-03-27 by I3lackshirts