MLB-LED-Scoreboard/mlb-led-scoreboard #319: Default pyowm package breaks on fresh install

jmordars, 2021-04-09:
Pyowm package in default install is not working, install a stable version until other ticket #310 is addressed

jmordars, 2021-04-09:
I don't have access to push to this repo -- the change I made was to remove pyowm in the sudo pip install, and instead create a new line
`sudo pip install git+https://github.com/csparpa/pyowm.git@1a29c198d797456e2fbb780aaf89582448490c74`

ty-porter, 2021-04-15:
@jmordars If this worked for you it might be a good idea to fork the repo and push to your local copy, then open a PR against the `dev` branch here?

jmordars, 2021-04-15:
Sure, you can do that off a fork?  Not familiar with pull requests off of forked repos. I'd rather not own a whole repository that differs from yours.

jmordars, 2021-04-15:
While I'm making this update, I'll make the pip change. That will affect anyone trying to work this until the python 3 updates go in

jmordars, 2021-04-15:
I'm new to GitHub. Is it simple to merge your master into mine if I fork?

jmordars, 2021-04-15:
I was hoping to get developer access to your repo, then I could make branches with changes. You can take or leave the changes via pull request.

swemoney, 2021-04-15:
That's not really how things things are usually done. It's not feasible for anyone with a change to get full access to the repo so you make a fork, do all of your changes, commit and push them and right in GitHub it'll prompt you to make a pull request. It gets reviewed and the maintainer can approve it or ask for changes from there. You can do this in separate branches if you want but in this repo, we usually ask that you derive from the `dev` branch.

Github has a lot of [documentation](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) to help you out..

swemoney, 2021-04-15:
also, to comment on your proposed change, I think it would be a lot cleaner to do something like `pip install pyowm==2.10.0 --force-reinstall` instead of picking a commit from a github repo.

swemoney, 2021-04-15:
I haven't tested that with the scoreboard so I'm not 100% that it'll work but I'd rather see a specific version being installed than grabbing a specific commit from github.

jmordars, 2021-04-15:
Ok, sounds good. I should have some time to work on some of this Friday.  I'm getting a covid shot, so if I'm feeling well I'll have time. 

 I worked a bit with @ty-porter on slack last week getting my board up and running, it's where we ran into these issues on my fresh install. I'm not too familiar with the GitHub workflow, but I'll give it a try. In past experience, doing a fork implies that I'm making divergent changes from what you all are doing.

swemoney, 2021-04-15:
Sure. Think of forks on a more small scale. There are large forks that take projects in completely new directions and stand on their own but I find those to be pretty rare. For the most part, in my experience, forks are used to create your own copy so you can create and test your branch and request it be added back to the main project. Think more about a fork in the road and less about the utensil with many prongs.

jmordars, 2021-04-15:
Ok, will give it a try.  What is the context between dev and master?

swemoney, 2021-04-15:
master is the branch that everyone checks out to update their board. dev is the branch we usually stage changes in before moving them to master so the more adventurous users have a chance to test things out before it goes out to everyone. So you'll just want to base your branch off the dev branch for your PR.

ty-porter, 2022-04-08:
Closed by #335

closed 2022-04-08 by ty-porter