MLB-LED-Scoreboard/mlb-led-scoreboard #336: MLB Scoreboard only displaying Weather data

richmyers, 2021-08-02:
Have two MLB scoreboards.  Power blip took one out.  Rebuilt...but only showing Weather data...which I have configured OpenWeather.  The other board was displaying upcoming game info.....THEN I restarted supervisor on the working board...and now it's only displaying Weather data.  Is the MLB feed down or do I have something wrong?  I have 2 teams and 2 divisions in my config.json file....so not everyone is on an off day.  Thanks in advance.

pjockey, 2021-08-04:
Are you using the version that uses Python2 or Python3?

Have your tried removing or commenting out one of your teams and re-running and see if there is any change?

ty-porter, 2021-08-06:
@richmyers Have you been able to reproduce this in the last couple days? If it's a one-off I always lean towards an MLB problem.

conorlastowka, 2022-03-19:
I'm having this same issue  where my board continued to display the weather even though a game was happening. Did you ever discover a fix?

twfarley, 2022-04-08:
This seems like a config issue. If you're still hitting this reach out on our Slack channel for help.

closed 2022-04-08 by twfarley