MLB-LED-Scoreboard/mlb-led-scoreboard #339: Board Not Moving Each Day

pjockey, 2021-08-20:
Wondering if I have something set wrong as I have to restart the board each day to get it to pick up the days games. It ends up looping the day before games.

```
{
	"preferred": {
		"teams": ["Blue Jays"],
		"divisions": ["AL East"]
	},
	"news_ticker": {
		"team_offday": false,
		"always_display": false,
		"preferred_teams": true,
		"traderumors": true,
		"mlb_news": true,
		"countdowns": true,
		"date": true,
		"date_format": "%A, %B %-d"
	},
	"standings": {
		"team_offday": true,
		"mlb_offday": true,
		"always_display": false,
		"display_no_games_live":false
	},
	"rotation": {
		"enabled": true,
		"scroll_until_finished": true,
		"only_preferred": false,
		"only_live": true,
		"rates": {
			"live": 15.0,
			"final": 15.0,
			"pregame": 15.0
		},
		"while_preferred_team_live": {
			"enabled": false,
			"during_inning_breaks": false
		}
	},
	"weather": {
                "team_offday": true,
		"apikey": "ef96cea0f9e9806c02181548e66671d2",
		"location": "Cranbrook, CA",
		"metric_units": true
	},
	"time_format": "24h",
	"end_of_day": "08:00",
	"full_team_names": true,
	"scrolling_speed": 3,
	"debug": false,
	"demo_date": false
}
```

twfarley, 2022-04-08:
let me know if you're still seeing this with the new update from today.

closed 2022-04-08 by twfarley

ChickenSupreme, 2022-04-12:
I'm seeing the same thing with the WardBrian fork which I installed about 2 weeks ago. I just updated to the main branch today so I'll report back if it's fixed or not. Here's the system log:

[log.txt](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/8476318/log.txt)

WardBrian, 2022-04-12:
This is definitely a bug I introduced. 

https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/b114e0f3038021d95d9e62be7287a8291a16d499/data/__init__.py#L23-L27

The code here only ever updates 'today' when the board is started, not later during refresh. 

I can fix this later today I think. My own board restarts itself each night so I never noticed.

ChickenSupreme, 2022-04-12:
For the record I'm pretty sure this might be related to the time sync thing someone else reported in the refactor thread as well. Looking more carefully at the log it does appear to be changing the clock forward dramatically (like 24h or more) after the scoreboard software starts. I added time-sync.target to my systemd unit file so that might help in the meantime.

Thanks for all your work on the scoreboard by the way, it looks great!