MLB-LED-Scoreboard/mlb-led-scoreboard #341: 'Failed to retrieve MLB important dates information' error

samcramer, 2021-11-16:
My scoreboard stopped working yesterday.  When I run main.py, I see the following:

$ sudo python main.py --led-gpio-mapping="adafruit-hat"
INFO (12:03:32): MLB LED Scoreboard - v4.0.1 (32x32)
Traceback (most recent call last):
  File "main.py", line 34, in <module>
    data = Data(config)
  File "/home/pi/src/scoreboard.latest/mlb-led-scoreboard/data/data.py", line 52, in __init__
    self.headlines = Headlines(self.config)
  File "/home/pi/src/scoreboard.latest/mlb-led-scoreboard/data/headlines.py", line 81, in __init__
    self.important_dates = Dates()
  File "/home/pi/src/scoreboard.latest/mlb-led-scoreboard/data/dates.py", line 8, in __init__
    self.__parse_important_dates(self.__fetch_important_dates())
  File "/home/pi/src/scoreboard.latest/mlb-led-scoreboard/data/dates.py", line 39, in __fetch_important_dates
    dates = mlbgame.important_dates()
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/__init__.py", line 234, in important_dates
    data = mlbgame.info.important_dates(year)
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/info.py", line 51, in important_dates
    data = mlbgame.data.get_important_dates(year)
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/data.py", line 52, in get_important_dates
    raise ValueError('Failed to retrieve MLB important dates information.')
ValueError: Failed to retrieve MLB important dates information.
$ 

Perhaps the API end point is not responding, but I'm not sure how to tell for sure.

ty-porter, 2021-11-16:
I think the API is down for a database migration:

```
http://lookup-service-prod.mlb.com/named.org_history.bam?org_id=1&season=2021
```

This returns:
```
Lookup Service Unavailable during database migration. Service expected to be restored by 11/18/2021.
```

This comes from the `mlbgame` library in the `get_important_dates` call.
https://github.com/panzarino/mlbgame/blob/master/mlbgame/data.py

samcramer, 2021-11-16:
Thanks!  I'll close this, since it's not a bug.

closed 2021-11-16 by samcramer