MLB-LED-Scoreboard/mlb-led-scoreboard #343: Board not showing game data

phonorad, 2022-03-23:
Is anyone having an issue with the board not showing game data, even though there are spring training games in progress?  I have 2 boards, both are showing only the weather page with MLB news scroll bar.  Someone posted what seems to be a similar problem in the previous issue which was posted Aug 2021.  Someone commented there 4 days ago that they are having the same problem as I am seeing.

pjockey, 2022-03-23:
There is another post in here over the past week

ty-porter, 2022-03-23:
Spring training games aren't shown in the API this board consumes. There isn't much we can do about it.

phonorad, 2022-03-23:
OK, thanks, I must have incorrectly remembered seeing spring training games on the board in past years.  Bring on the Regular Season!

ty-porter, 2022-03-23:
It might have worked in the past, but we don't control that API. That's why there's a disclaimer in the README 😃

phonorad, 2022-03-23:
Gotcha, no problem, just wanted to make sure that something had not broken!  Thanks for the info and your efforts!

closed 2022-03-23 by phonorad