MLB-LED-Scoreboard/mlb-led-scoreboard #344: MLBGAME not producing ... anything

pfightingpolish, 2022-03-28:
Soooooo not to be the bearer of bad news or anything, but while I know we closed the issue about spring training games not loading, I've been trying to use `mlbgame` with the `demo_date` functionality to load past regular-season games and haven't been getting anything, either. I tried both through this project's (`MLB-LED-Scoreboard`) functionality and through command-line python and got nada. I've even tried hitting results just through a browser and have just been hitting 404 errors. 

Trust me when I say this once used to work. I actually remember testing scores from 1957 through the board and getting data in the past. Now, I'm not even getting data from 2021. 

I hate to say it, but we've talked in the past about MLB gradually phasing out usage of the API behind `mlbgame`. I am wondering if it's not merely a spring training issue but if `mlbgame` has just bit the dust, period. Given what MLB did to eliminate any and all references to current MLBPA members during the lockout, it wouldn't totally surprise me. 

I am wondering if this might not be the time to simply bite the bullet and try to write `mlbgame` out of the code in favor of the much more up-to-date `statsapi`. I have noticed `statsapi` is producing spring training game data, but I'm still tinkering with it to try and get a daily schedule. 

Writing out `mlbgame` has been discussed here for some time. While I won't say `mlbgame` might not just pop back to life at some point, if it doesn't, it will have to happen, anyway, if we want the project to live. 

An alternate solution could also be to switch to ESPN's API. For the boards I've created for other sports, I've used that API extensively. However, I think they pull their data from MLB anyhow, so I think there would be greater lag with that. I think we might as well just go to the source. 

I know it's not a fun prospect, as it means we'll both have to familiarize ourself with `statsapi` and rewrite a healthy amount of the guts of the data directory. But I think it might just be time, for better or worse. 

I'll start looking at it.

pjockey, 2022-03-28:
Have you looked at this fork? https://github.com/WardBrian/mlb-led-scoreboard

I had preseason game on my board yesterday. It is python 3 rewrite.

pfightingpolish, 2022-03-28:
I'll take a look. 

I've customized my board a lot, so I've been trying to avoid major refactors because my own personal fork has gone off in a very different direction. But I think this may force the issue.

WashCaps33, 2022-03-28:
For those of us using the MLB API, what would we need to do to switch over to the statsapi?  I have barely any experience with coding so I am hoping this is not something that will essentially be the death of my board.

WardBrian, 2022-04-08:
Seems like MLBGame also didn’t get opening day data, which prompted my fork to get merged

twfarley, 2022-04-08:
Yep. Fixed with new pull.

closed 2022-04-08 by twfarley