MLB-LED-Scoreboard/mlb-led-scoreboard #348: After update

richmyers, 2022-04-11:
after running ./install.sh to get updates....this is what occurs when attempting to run.

pi@MLBScoreboard:/etc/supervisor/conf.d $ sudo python3 /home/pi/mlb-led-scoreboard/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64 --led-no-hardware-pulse=NO_LED_HARDWARE_PULSE
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 17, in <module>
    from renderers.main import MainRenderer
ImportError: bad magic number in 'renderers': b'\x03\xf3\r\n'

ty-porter, 2022-04-11:
Likely due to compiled Python files (`.pyc`) that need to be removed:
https://mlb-led-scoreboard.slack.com/archives/C9BLJ9JF9/p1649449813778739

jbark25, 2022-04-11:
Hello, I have a similar issue and would like to access the SLACK to see your solution. Any way i can get an invite? jeffkbarker@gmail.com

ty-porter, 2022-04-11:
It looks like our README link was accidentally removed with the latest
update, you can request an invite here:

https://mlb-led-scoreboard.herokuapp.com/

Thanks,
Tyler Porter

twfarley, 2022-04-13:
> Likely due to compiled Python files (`.pyc`) that need to be removed:
> https://mlb-led-scoreboard.slack.com/archives/C9BLJ9JF9/p1649449813778739

Let us know if this didn't resolve the issue for you.

closed 2022-04-13 by twfarley