MLB-LED-Scoreboard/mlb-led-scoreboard #349: Syntax error in data/__init__.py

samcramer, 2022-04-11:
After doing a fresh `git pull` of `https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard.git` and running `sudo ./install.sh` I am seeing this as part of the installation process

``` Using cached https://files.pythonhosted.org/packages/88/99/503196ca5e9e69a44d1b3e0e1bc0ca2a2fa0c12b09041b14b52bb3ef3523/feedparser-6.0.0-py2.py3-none-any.whl
Collecting pyowm>3
  Could not find a version that satisfies the requirement pyowm>3 (from versions: 0.2.0, 0.4.0, 0.4.1, 1.0.0, 1.2.0, 2.0.0, 2.2.0, 2.2.1, 2.3.0, 2.3.1, 2.3.2, 2.4.0, 2.5.0, 2.6.0, 2.6.1, 2.7.1, 2.8.0, 2.9.0, 2.10.0)
No matching distribution found for pyowm>3
```

When I run `sudo python3 main.py` I see this crash:
```
Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from data import Data
  File "/home/pi/src/mlb-led-scoreboard-04112022/data/__init__.py", line 27
    self.schedule: Schedule = Schedule(self.today, config)
                 ^
SyntaxError: invalid syntax
```

samcramer, 2022-04-11:
Here's my python version:
```
$ python --version
Python 3.5.3
$
```

ty-porter, 2022-04-11:
`pyowm>3` for sure requires Python 3.7 or higher. The typing error you ran into at the bottom likely is resolved by higher Python version as well.

Easiest way to do that is probably to upgrade PiOS unfortunately

samcramer, 2022-04-11:
Upgrading the OS fixed it.

closed 2022-04-11 by samcramer