MLB-LED-Scoreboard/mlb-led-scoreboard #350: Exclamation point in lower right corner

samcramer, 2022-04-11:
I'm sorry if this isn't the right place to ask this question, but I'm not aware of a forum of mailing list that I should use instead.

There's an exclamation point in the lower right corner of my display.  What does it mean?

ty-porter, 2022-04-11:
There is a Slack channel that might be helpful: https://mlb-led-scoreboard.herokuapp.com/

There's also a display for network issues on some API calls that this might be related to

WardBrian, 2022-04-12:
This means one of the network calls on the board failed. If you have debug enabled in your config, the output of the program should log which one is failing

twfarley, 2022-04-13:
set `"debug": true` in the config.json, then run the scoreboard by calling `sudo python3 main.py --led-gpio-mapping="adafruit-hat"` plus whatever other arguments you usually run with. Running as a service doesn't show any logs. If the issue reproduces we should be able to help if you provide the error message you see.

Noah-Brown, 2022-04-13:
I am getting the same issue occasionally, Which Raspberry Pi are you using? I am using a pi zero W. Which I suspect is the root of my problem.

samcramer, 2022-04-13:
I'm not sure if the question was directed at me, but I'm using a Raspberry Pi 4 model B.

twfarley, 2022-04-13:
> I am getting the same issue occasionally, Which Raspberry Pi are you using? I am using a pi zero W. Which I suspect is the root of my problem.

This was resolved by using dietPi.

Electromag1888, 2022-10-23:
In raspi-config, the option -network connection needs to be made before boot up.

WardBrian, 2025-03-02:
Closing old issues, please open a new one or start a discussion if running into more set-up issues

closed 2025-03-02 by WardBrian