MLB-LED-Scoreboard/mlb-led-scoreboard #352: Lower half of board.. dim to black

richmyers, 2022-04-13:
After update the lower half of the board..is not quite black. The file in the ledcolors directory is set to background 0.0.0....  This worked perfectly prior to update.

twfarley, 2022-04-13:
Could you share the contents of your scoreboard.json? Also, with the latest update the custom colors and layouts have moved to /colors/ and /coordinates/ directories. You might have to update those examples to your liking.

richmyers, 2022-04-14:
{
  "bases": {
    "1B": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "2B": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "3B": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "final": {
    "inning": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "scrolling_text": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "nohit_text": {
      "r": 255,
      "g": 50,
      "b": 50
    }
  },
  "inning": {
    "break": {
      "number": {
        "r": 255,
        "g": 235,
        "b": 59
      },
      "text": {
        "r": 255,
        "g": 235,
        "b": 59
      }
    },
"number": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "arrow": {
      "up": {
        "r": 255,
        "g": 235,
        "b": 59
      },
      "down": {
        "r": 255,
        "g": 235,
        "b": 59
      }
    }
  },
  "outs": {
    "1": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "2": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "3": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "batter_count": {
    "r": 255,
    "g": 235,
    "b": 59
  },
  "nohit_text": {
    "r": 255,
    "g": 110,
    "b": 110
  },
 "perfect_game_text": {
    "r": 255,
    "g": 110,
    "b": 110
  },
  "pregame": {
    "matchup": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "scrolling_text": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "start_time": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "warmup_text": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "status": {
    "text": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "scrolling_text": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "standings": {
    "background": {
      "r": 37,
      "g": 102,
      "b": 30
    },
    "divider": {
      "r": 13,
      "g": 35,
      "b": 11
    },
    "stat": {
      "r": 171,
      "g": 181,
      "b": 170
    },
    "team": {
      "name": {
        "r": 171,
        "g": 181,
        "b": 170
      },
      "stat": {
        "r": 171,
        "g": 181,
        "b": 170
      }
    }
  },
  "offday": {
    "scrolling_text": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "time": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "conditions": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "temperature": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "wind_speed": {
      "r": 255,
      "g": 235,
      "b": 59
    },
   },
    "wind_dir": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "wind": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "weather_icon": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "network": {
    "background": {
      "r": 255,
      "g": 0,
      "b": 0
    },
    "text": {
      "r": 255,
      "g": 255,
      "b": 255
    }
  },
  "default": {
    "background": {
      "r": 0,
      "g": 0,
      "b": 0
    },
    "text": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  }
}

richmyers, 2022-04-14:
ok...maybe I got it.  I found another scoreboard in the /colors directory.  Coped that to .json and edited the background.  Will see if that corrects the issue.  Thanks much for the pointer

closed 2022-04-14 by twfarley