MLB-LED-Scoreboard/mlb-led-scoreboard #361: Switching Modes

rhazzed, 2022-04-17:
I am new to MLB app and having difficulty in configuration.  Seems the board is an either or design.  
What I am trying to do is have the weather/news display until preferred team starts and it seems impossible to do.  I have news displayed always on... but then I cant view a game and visa versa.  if set to false, I cant see the news or weather. Am I misconfiguring something or is this board simply not capable of this?  Hard to believe this would be an intentional design to be forced to only watch one thing or another.  Have a friend with similar issue. 

Please help.

WardBrian, 2022-04-17:
What you describe is true, you can’t mix weather or standings into the games rotation. The exception is you are able to display standings when no games are live. 

The way that the program is currently set up makes it difficult to allow the level of customization to do what you describe. It would be nice, it’s just difficult to implement based on how it currently works

rhazzed, 2022-04-17:
So the option is to only see weather/news on game/mlb off days where there
is no rotation to contend with?

twfarley, 2022-04-18:
Correct - with the way the board works now is  that you can see the weather/offday screen when your preferred team has an off day with `"team_offday": true`, or always with `"always_display": true`.

However, I've created two config files that I keep in the home directory called configDay and configNight. The day version has `always_display` set to false while the night has `always_display` set to true. I set a cron job to overwrite the existing in the mlb-led-scoreboard directory and restart the service when I want the news/time to display.

I can create a wiki for this if you want more details.

Closing this as we don't plan on implementing this feature.

closed 2022-04-18 by twfarley