MLB-LED-Scoreboard/mlb-led-scoreboard #365: Integrate broadcast service into pregame info

twfarley, 2022-04-19:
https://statsapi.mlb.com/api/v1/schedule?sportId=1&date=2022-04-10&hydrate=broadcasts

WardBrian, 2022-04-19:
There will be two difficulties with this:

One, the schedule is one of the few places we call a function in statsapi rather than using the endpoint directly. So, we’d probably want to add this to the schedule function there, see: https://github.com/toddrob99/MLB-StatsAPI/issues/71

Secondly, this is not available from the game endpoint, so it will need to be passed in from the constructor. This isn’t a conceptually difficult thing to do, but it’s unfortunate from a standpoint of code cleanliness

WardBrian, 2022-08-04:
I have a working branch that does this here: https://github.com/WardBrian/mlb-led-scoreboard/tree/pregame-broadcast

It depends on https://github.com/toddrob99/MLB-StatsAPI/pull/85

ZachPL, 2022-08-04:
I'm going to try this out soon, thanks

WardBrian, 2022-08-04:
For now I just stuck it at the end of the starting pitchers
[image]

ZachPL, 2022-08-09:
@WardBrian I commented out the is isnational check, in order for it to show for example YES network for the yankees, it might not be ideal for it to show all broadcasts including radio stations espeicially for everyone, but maybe a parameter to turn that on or off?

This year the yankees can be on prime, youtube, yes, national broadcasts, so it's pretty convenient to show them all.

WardBrian, 2022-08-09:
My theory is if it doesn’t say anything, you should be able to know it’s the default (E.g all of those besides YES should be marked as national)

ZachPL, 2022-08-09:
That makes sense, for my use case my grandpa texts me every day what channel the game is on and I know if it doesn't say specifically he will have questions.

PR #403 (2022-08-15): Show national broadcasts during pregame
This closes #365 

[image]
closed 2023-02-28 by WardBrian