MLB-LED-Scoreboard/mlb-led-scoreboard #368: Working / Quit working

richmyers, 2022-04-23:
Scoreboard working fine the exited and will not restart.   Below is the error I get when start from cmd line.    Not sure why script is looking for w32h32 file when I have w32h64.json configured to blackout lower half of scoreboard.   Always....appreciate the support.

pi@MLScoreboards:~/mlb-led-scoreboard $ cd coordinates/
pi@MLScoreboards:~/mlb-led-scoreboard/coordinates $ ls
__init__.py  w128h32.json.example  w32h32.json.example  w64h32.json.example
README.md    w128h64.json.example  w64h32.json          w64h64.json.example
pi@MLScoreboards:~/mlb-led-scoreboard/coordinates $ sudo python3 /home/pi/mlb-led-scoreboard/main.py --led-gpio-mapping=adafruit-hat
WARNING (08:46:29): Could not find json file config.json.example.  Skipping.
WARNING (08:46:29): Could not find json file config.json.  Skipping.
ERROR (08:46:29): Untrapped error in main!
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 144, in <module>
    main(matrix, config)
  File "/home/pi/mlb-led-scoreboard/main.py", line 37, in main
    config = Config(config_base, matrix.width, matrix.height)
  File "/home/pi/mlb-led-scoreboard/data/config/__init__.py", line 25, in __init__
    self.preferred_teams = json["preferred"]["teams"]
KeyError: 'preferred'
pi@MLScoreboards:~/mlb-led-scoreboard/coordinates $ cd ,,
-bash: cd: ,,: No such file or directory
pi@MLScoreboards:~/mlb-led-scoreboard/coordinates $ cd ..
pi@MLScoreboards:~/mlb-led-scoreboard $ ls
assets       config.json.bu       coordinates  debug.pyc     ledcolors   main.py      pyproject.toml  submodules
colors       config.json.example  data         highlight.py  ledcoords   matrix       README.md       utils.py
config.json  CONTRIBUTING.md      debug.py     install.sh    LICENSE.md  __pycache__  renderers       utils.pyc
pi@MLScoreboards:~/mlb-led-scoreboard $ more config.json
{
        "preferred": {
                "teams": ["Royals"],
                "divisions": ["AL Central"]
        },

ty-porter, 2022-04-23:
You're not in the right directory. Everything is loaded relatively. In your case you need to be in the `home/pi/mlb-led-scoreboard` directory.

```
cd ./home/pi/mlb-led-scoreboard
python3 main.py --led-gpio-mapping=adafruit-hat
```

If you want to use 64x32, you need to specify the columns:

```
python3 main.py --led-cols=64 main.py --led-gpio-mapping=adafruit-hat
```

---

If this still doesn't work, there is another issue that the stack trace you posted didn't cover.

richmyers, 2022-04-23:
yeah....sorry....this is the entire command...as specified in the Supervisor config file

sudo python3 /home/pi/mlb-led-scoreboard/main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64

This is the output after copying the config.json.example to config.json.  No other changes made.

pi@MLScoreboards:~/mlb-led-scoreboard $ sudo python3 /home/pi/mlb-led-scoreboard/main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2 --led-ro    ws=32 --led-cols=64
WARNING (09:15:53): [WEATHER] The API key provided doesn't appear to be valid. Please check your config.json.
WARNING (09:15:53): [WEATHER] You can get a free API key by visiting https://home.openweathermap.org/users/sign_up
ERROR (09:15:56): Untrapped error in main!
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 144, in <module>
    main(matrix, config)
  File "/home/pi/mlb-led-scoreboard/main.py", line 65, in main
    data = Data(config)
  File "/home/pi/mlb-led-scoreboard/data/__init__.py", line 36, in __init__
    self.headlines: Headlines = Headlines(config, self.today.year)
  File "/home/pi/mlb-led-scoreboard/data/headlines.py", line 107, in __init__
    self.update(True)
  File "/home/pi/mlb-led-scoreboard/data/headlines.py", line 122, in update
    f = feedparser.parse(url)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/api.py", line 214, in parse
    data = _open_resource(url_file_stream_or_string, etag, modified, agent, referrer, handlers, request_headers, result)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/api.py", line 114, in _open_resource
    return http.get(url_file_stream_or_string, etag, modified, agent, referrer, handlers, request_headers, result)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/http.py", line 158, in get
    f = opener.open(request)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 563, in error
    result = self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/http.py", line 55, in http_error_301
    result = urllib.request.HTTPRedirectHandler.http_error_301(self, req, fp, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 755, in http_error_302
    return self.parent.open(new, timeout=req.timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 563, in error
    result = self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/http.py", line 55, in http_error_301
    result = urllib.request.HTTPRedirectHandler.http_error_301(self, req, fp, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 755, in http_error_302
    return self.parent.open(new, timeout=req.timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 563, in error
    result = self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/http.py", line 55, in http_error_301
    result = urllib.request.HTTPRedirectHandler.http_error_301(self, req, fp, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 755, in http_error_302
    return self.parent.open(new, timeout=req.timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 563, in error
    result = self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/http.py", line 55, in http_error_301
    result = urllib.request.HTTPRedirectHandler.http_error_301(self, req, fp, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 755, in http_error_302
    return self.parent.open(new, timeout=req.timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 563, in error
    result = self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/local/lib/python3.7/dist-packages/feedparser/http.py", line 55, in http_error_301
    result = urllib.request.HTTPRedirectHandler.http_error_301(self, req, fp, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 745, in http_error_302
    self.inf_msg + msg, headers, fp)
urllib.error.HTTPError: HTTP Error 301: The HTTP server returned a redirect error that would lead to an infinite loop.
The last 30x error message was:
Moved Permanently

richmyers, 2022-04-23:
pi@MLScoreboards:~/mlb-led-scoreboard $ ls
assets  config.json     config.json.example  coordinates  debug.py      install.sh  ledcoords   main.py  __pycache__     README.md  submodules
colors  config.json.bu  CONTRIBUTING.md      data         highlight.py  ledcolors   LICENSE.md  matrix   pyproject.toml  renderers  utils.py
pi@MLScoreboards:~/mlb-led-scoreboard $ ls -al
total 160
drwxr-xr-x 13 root root  4096 Apr 23 09:16 .
drwxr-xr-x  9 pi   pi    4096 Apr 15 07:41 ..
drwxr-xr-x  4 root root  4096 Apr 11 14:42 assets
drwxr-xr-x  2 root root  4096 Apr 23 08:45 colors
-rwxr-xr-x  1 root root   947 Apr 23 09:14 config.json
-rw-r--r--  1 root pi     888 Apr 15 08:12 config.json.bu
-rw-r--r--  1 root root   947 Apr 23 08:55 config.json.example
-rw-r--r--  1 root root   168 May 21  2021 CONTRIBUTING.md
drwxr-xr-x  2 root root  4096 Apr 23 08:32 coordinates
drwxr-xr-x  5 root root  4096 Apr 11 14:44 data
-rw-r--r--  1 root root   355 Apr 11 14:42 debug.py
-rw-r--r--  1 root root    69 Apr 11 14:42 .flake8
drwxr-xr-x  9 root root  4096 Apr 11 14:42 .git
-rw-r--r--  1 root root  1453 Apr 11 14:42 .gitignore
-rw-r--r--  1 root root  1146 Apr 11 14:42 highlight.py
-rwxr-xr-x  1 root root   499 Apr 11 14:42 install.sh
drwxr-xr-x  2 root root  4096 Apr 11 14:42 ledcolors
drwxr-xr-x  2 root root  4096 Apr 11 14:42 ledcoords
-rw-r--r--  1 root root 35147 May 21  2021 LICENSE.md
-rwxr-xr-x  1 root root  4441 Apr 11 14:42 main.py
drwxr-xr-x 10 root root  4096 May 21  2021 matrix
drwxr-xr-x  2 root root  4096 Apr 11 15:06 __pycache__
-rw-r--r--  1 root root    30 Apr 11 14:42 pyproject.toml
-rwxr-xr-x  1 root root 16079 Apr 11 14:42 README.md
drwxr-xr-x  4 root root  4096 Apr 11 15:06 renderers
drwxr-xr-x  3 root root  4096 Apr 11 14:51 submodules
-rw-r--r--  1 root root  6555 Apr 11 14:42 utils.py

ty-porter, 2022-04-23:
This looks related to the news feed:

```
The last 30x error message was:
Moved Permanently
```

301 is `Moved Permanently` and looks like one of the feeds is redirecting in a loop.

I'm not able to reproduce it locally, however you should be able to turn off both MLB and TradeRumor feeds in config. If one of those resolves the issue it would help narrow down which of the feed URLs needs to be updated. I'm guessing this is Indians -> Guardians related but unsure at the moment.

ty-porter, 2022-04-25:
@richmyers did this resolve over the weekend or is it still an issue?

closed 2022-05-23 by twfarley