MLB-LED-Scoreboard/mlb-led-scoreboard #373: Cant seem to get it to work

Slythy, 2022-04-28:
I followed the guide. have a adafruit bonnet and a 64x32 led. If i put everything in the /home/pi/ folder i can get it to boot up and display something but its clearly displaying bad text. looks like its setup for a 32x32. ive read the read me's but I figured i shouldnt have to type the /home/pi/mlb-led to get it to work every time right?

Slythy, 2022-04-28:
[scoreboard picture](https://imgur.com/gallery/e81tovY)

is a picture of what its currently doing

Slythy, 2022-04-28:
Ok, so setting led-slow-down-gpio4 seemed to get the screen to display. Next issue is I cant for the life of me get it to display as a 64x32. Its replicating a 32x32 matrix on both sides. [seen here](https://i.imgur.com/wN9U0u7.jpeg).

I set the example 64x32 json example to just a json. is there anything else I have to do? Is there something physical I need to change on my bonnet?

 Is there a list of teams available? I just want to make sure I type them in correctly

lastly is there a way to run a code on startup so I dont have to type it in when I start it up so it just auto startups?

ty-porter, 2022-04-29:
Closing, issue was resolved in Slack and will reopen on reoccurrence.

closed 2022-04-29 by ty-porter