MLB-LED-Scoreboard/mlb-led-scoreboard #387: Seeing shortened team names with RHE off

twfarley, 2022-05-23:
```
   "runs": {
      "runs_hits_errors": {
        "show": false,
        "compress_digits": false,
        "spacing": 3
      },  
```          

I'm still seeing shortened team names with this disabled.

WardBrian, 2022-08-04:
Do you have `"full_team_names": false,` in config.json?

twfarley, 2022-08-04:
No, it's set to true.

twfarley, 2022-08-04:
and in coords file
```
    "runs": {
      "runs_hits_errors": {
        "show": false,
        "compress_digits": false,
        "spacing": 3
      }, 
```

ty-porter, 2023-02-25:
This can be repro'd by always setting `runs` to > 9 in `data/scoreboard/team.py` while a game is ongoing

```python
class Team:
    def __init__(self, abbrev, runs, name, hits, errors):
        self.abbrev = abbrev
        self.runs = 10 
        self.name = name
        self.hits = hits
        self.errors = errors
```

twfarley, 2023-02-26:
I didn't have `"short_team_names_for_runs_hits": true,` I my config file...

No longer reproducing after adding it. /facepalm

WardBrian, 2023-02-27:
I think we can close this in favor of #422 then

closed 2023-02-27 by WardBrian