MLB-LED-Scoreboard/mlb-led-scoreboard #389: Failed on start

eeedlef, 2022-06-20:
## This issue is a

- [ X ] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi 4B
- Operating System: Debian
- Matrix size: 64x32
- Display adapter (HAT/Bonnet/etc.): Bonnet
  - Additional information:
- Power source(s): 

## Software Configuration

- MLB LED Scoreboard Version: v5.1.1
- Commands/arguments used to start the scoreboard: $ sudo service mlb-led-scoreboard start

## Expected behavior
Startup of mlb-led-scoreboard

## Actual behavior
Only two leds are lit, and the scoreboard does not otherwise load

## Additional Information

From the log:

mlb-led-scoreboard.service: Changing to the requested working directory failed: No such file or directory
mlb-led-scoreboard.service: Failed at step CHDIR spawning /usr/bin/python3: No such file or directory
mlb-led-scoreboard.service: Main process exited, code=exited, status=220/CHDIR
mlb-led-scoreboard.service: Failed with result 'exit-code'.

WardBrian, 2022-06-20:
Do you have python 3 installed?

eeedlef, 2022-06-20:
Yes. I think I discovered that it is related to the sound causing an issue. I've disabled using the tip here: https://forums.raspberrypi.com/viewtopic.php?t=260568

But when I completed that step, it now still has an issue with the "no-hardware-pulse" argument:

usage: main.py [-h] [--led-rows LED_ROWS]  [--led-cols LED-COLS]
[--led-chain LED_CHAIN]  [--led-parallel LED_PARALLEL]
[--led-pwm-bits LED_PWM_BITS]  [--led-brightness LED_BRIGHTNESS]
[--led-gpio-mapping {regular,adafruit-hat,adafruit-hat-pwm}]
[--led-scan-mode {0,1}]
[--led-pwm-lsb-nanoseconds LED_PWM_LSB_NANOSECONDS]
[--led-no-hardware-pulse LED_NO_HARDWARE_PULSE]
[--led-rgb-sequence LED_RGB_SEQUENCE]
[--led-pixel-mapper LED_PIXEL_MAPPER]
[--led-row-addr-type {0,1,2,3}]
[--led-multiplexing LED_MULTIPLEXING]
[--led-limit-refresh LED_LIMIT_REFRESH]
[--led-pwm-dither-bits LED_PWM_DITHER_BITS]  [--config CONFIG]
main.py: error: argument --led-no-hardward-pulse: expected one argument

eeedlef, 2022-06-20:
I'm embarrassed, I just found the sound issue discussion in your read me. I'm working through those steps currently...