MLB-LED-Scoreboard/mlb-led-scoreboard #399: Introduce City Connect colors

twfarley, 2022-08-03:
## This issue is a

- [ ] Bug Report
- [x] Feature Request

Introduce behavior to use applicable City Connect colors on Tuesdays for home games.

WardBrian, 2022-08-03:
I think different teams do city connects on different days. Nationals do it on Saturday

twfarley, 2022-08-03:
Ah. Well, that makes this a lot more difficult. Unless we can pull that from stats. But I doubt it.

WardBrian, 2022-08-04:
https://statsapi.mlb.com/api/v1/schedule?sportId=1&gamePk=662476&hydrate=game(promotions)

They appear to be listed as "promotions" under the schedule endpoint if you hydrate `game(promotions)`

Doesn't seem like there's a super standard format for this though, see tonight's angels game for another:
https://statsapi.mlb.com/api/v1/schedule?sportId=1&gamePk=663375&hydrate=game(promotions)

gotopes, 2025-02-24:
this could actually become a reality now.  a team's jersey in a game is now in the API so the program would  need is ask the API what jersey the team is wearing in a particular game and if it comes back as their cityconnect, then the scoreboard part can be colorized, imaged, or screenwrite text that fact.

WardBrian, 2025-02-24:
Do you know how it shows up in the api exactly?

gotopes, 2025-02-26:
here's an example with the Rockies wearing cityconnects in that game

https://statsapi.mlb.com/api/v1/uniforms/game/?gamePks=746508

PR #578 (2025-04-05): Special uniform color support
Closes #399 (city connects). I tried to write this in a general enough way that we could extend it to e.g. throwback or other special uniforms if we want to

Depends on https://github.com/toddrob99/MLB-StatsAPI/pull/154

Draft status until the above is released.

Here's what the [double city connect game](https://www.foxsports.com/stories/mlb/nationals-orioles-will-be-first-teams-to-wear-city-connect-jerseys-against-each-other) looks like:

[image]
closed 2025-04-20 by ty-porter