MLB-LED-Scoreboard/mlb-led-scoreboard #400: On startup, initializes to weather and stays there...

mmendelson222, 2022-08-03:
## This issue is a

- [ ] Bug Report
- [ ] Feature Request
- [x] Exploratory question

## Behavior

When I start up 
[mlb.log](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/9253389/mlb.log)
my pi, for some reason the connection is not immediately available.  The app initializes to "weather" and stays there. 

A reboot solves the problem, but this presents obvious problems when running headless.  
[config-all.json.txt](
[config-all.json.txt](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/9253391/config-all.json.txt)
https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/9253386/config-all.json.txt)

Before I assume that this is a problem with the app, is there some way to work around this issue?  Perhaps there's a way to increase the timeout, which I'm missing. 

I'll attach my configuration and the debug log. 

Thanks.

ty-porter, 2022-08-03:
You should verify your startup script looks like the one that is in the wiki: https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/wiki

This isn't really an issue with the scoreboard itself, more that the Pi isn't waiting for a connection before running your startup command.

Secondly, the config you uploaded has an API key in it. You should remove the file from the issue as it could be compromised.

mmendelson222, 2022-08-03:
Thanks @ty-porter, very helpful.

closed 2022-08-03 by mmendelson222