MLB-LED-Scoreboard/mlb-led-scoreboard #402: [Feature Request] Add minor league teams/games.

ForceConstant, 2022-08-08:
## This issue is a

- [ ] Bug Report
- [x] Feature Request

Would be great if I could choose minor league teams also to display. They are accessed by using "#  sportId=11 for AAA, 12 for AA, 13 for A+, and 14 for A."

Dbeaver2314, 2023-04-15:
Played around with this tonight. It will be a pretty large undertaking but here is proof it can be done.
The API will allow us to do all pro teams (AAA down to A) and college games too.

So many features break when you set this up however. Things like the news feed rely on having an MLB Major League in your config file.
all of these teams will need colors manually defined as well.
The standings file will need a lot more cases, right now its only setup for AL and NL.
Probably many many more steps needed. I would love to keep working on this, but we should probably look at opening an epic if we want this to happen

[image]

twfarley, 2023-04-15:
This might work better as a fork than directly in the same repo.

closed 2026-09-21 by twfarley