MLB-LED-Scoreboard/mlb-led-scoreboard #404: Error at start with KeyError: 'national_broadcasts'

smith288, 2022-08-19:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Software Configuration

- MLB LED Scoreboard Version: v5.1.1
- Commands/arguments used to start the scoreboard:

## Expected behavior
Proceed to scoreboard

## Actual behavior
Crash

## Additional Information
Error 
---
DEBUG (16:00:51): Updating schedule for 2022-08-19 16:00:51.227222
ERROR (16:00:51): Untrapped error in main!
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 141, in <module>
    main(matrix, config)
  File "/home/pi/mlb-led-scoreboard/main.py", line 62, in main
    data = Data(config)
  File "/home/pi/mlb-led-scoreboard/data/__init__.py", line 25, in __init__
    self.current_game: Game = self.schedule.get_preferred_game()
  File "/home/pi/mlb-led-scoreboard/data/schedule.py", line 93, in get_preferred_game
    return self.__current_game()
  File "/home/pi/mlb-led-scoreboard/data/schedule.py", line 149, in __current_game
    return Game.from_ID(scheduled_game["game_id"], self.date, scheduled_game["national_broadcasts"])
KeyError: 'national_broadcasts'

ty-porter, 2022-08-19:
Is this on dev?

smith288, 2022-08-21:
git branch says main

WardBrian, 2022-08-21:
Main is probably main of my fork then. You will need to update mlb-statsapi to version 1.5

smith288, 2022-08-21:
Oh snap… you're right. My bad. I'll close. Thanks Brian.

closed 2022-08-21 by smith288