MLB-LED-Scoreboard/mlb-led-scoreboard #408: Network error indicator misaligned

feram18, 2022-09-02:
## This issue is a

- [X] Bug Report
- [ ] Feature Request

## Software Configuration

- MLB LED Scoreboard Version: `5.1.4`

## Expected behavior

Network error indicator should be displayed in the bottom right corner

## Actual behavior

- `128x32` layout has exclamation mark misaligned
- `128x64` layout has both the background and text misaligned

ty-porter, 2022-09-03:
Screenshots of the issue:

128x32:
[image]

128x64
[image]

ty-porter, 2022-09-03:
Addressed by #409

closed 2022-09-03 by ty-porter