MLB-LED-Scoreboard/mlb-led-scoreboard #410: IndexError: list index out of range (crash)

kennedye, 2022-09-18:
## This issue is a

- [X ] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): `2B v1.1`
- Operating System: `Raspian OS Bullseye 11.4`
- Matrix size: `64x32`
- Display adapter (HAT/Bonnet/etc.): `Adafruit RGB Matrix bonnet (no PWM mod)`
  - Additional information:
- Power source(s): `Adafruit 5V 4A power supply (powering the Pi from the Bonnet and not separately)`

## Software Configuration

- MLB LED Scoreboard Version: `MLB LED Scoreboard v5.1.1`
- Commands/arguments used to start the scoreboard: `sudo python3 main.py --led-gpio-mapping="adafruit-hat" --led-rows=32 --led-cols=64 --led-slowdown-gpio=2 --led-brightness=50 --led-show-refresh`

## Expected behavior
runs as expected

## Actual behavior
occasionally (not always, usually only seems to happen if I leave it running for like 2-3 hours) crashes with the error listed below, same place every time too.

## Additional Information
```
ERROR (13:15:19): Untrapped error in main!
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 141, in <module>
    main(matrix, config)
  File "/home/pi/mlb-led-scoreboard/main.py", line 75, in main
    __refresh_games(render, data)
  File "/home/pi/mlb-led-scoreboard/main.py", line 125, in __refresh_games
    data.advance_to_next_game()
  File "/home/pi/mlb-led-scoreboard/data/__init__.py", line 75, in advance_to_next_game
    game = self.schedule.next_game()
  File "/home/pi/mlb-led-scoreboard/data/schedule.py", line 106, in next_game
    preferred_game = Game.from_ID(self._games[game_index]["game_id"], self.date)
IndexError: list index out of range
```

WardBrian, 2022-09-18:
Can you post the config.json you are using?

kennedye, 2022-09-20:
here 'tis:
```
{
	"preferred": {
		"teams": ["Mariners","Nationals","Orioles"],
		"divisions": ["AL West", "NL East", "AL Wild Card"]
	},
	"news_ticker": {
		"team_offday": true,
		"always_display": false,
		"preferred_teams": true,
		"traderumors": false,
		"mlb_news": false,
		"countdowns": true,
		"date": true,
		"date_format": "%A, %B %-d"
	},
	"standings": {
		"team_offday": false,
		"mlb_offday": false,
		"always_display": false,
		"display_no_games_live": false
	},
	"rotation": {
		"enabled": true,
		"scroll_until_finished": true,
		"only_preferred": false,
		"only_live": true,
		"rates": {
			"live": 15.0,
			"final": 15.0,
			"pregame": 15.0
		},
		"while_preferred_team_live": {
			"enabled": false,
			"during_inning_breaks": true
		}
	},
	"weather": {
		"apikey": "<redacted>",
		"location": "Seattle,wa,us",
		"metric_units": false
	},
	"time_format": "12h",
	"end_of_day": "00:00",
	"full_team_names": true,
	"short_team_names_for_runs_hits": true,
	"pregame_weather": true,
	"scrolling_speed": 2,
	"debug": false,
	"demo_date": false
}
```

ty-porter, 2023-09-29:
Closing, looks like #455 cleared this up.

closed 2023-09-29 by ty-porter