MLB-LED-Scoreboard/mlb-led-scoreboard #413: Can't find mlb news feeds

pjockey, 2022-10-04:
## This issue is a

- [ ] Bug Report
~/mlb-led-scoreboard $ sudo python3 main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-cols=128 --led-rows=64 --led-rgb-sequence=BGR --led-row-addr-type=3
WARNING (20:39:08): There was a problem fetching https://www.mlb.com//feeds/news/rss.xml

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): 3B+
- Operating System: Buster
- Matrix size: 128x64
- Display adapter (HAT/Bonnet/etc.): Bonnet
  - Additional information:
- Power source(s): RPI official power adapter

## Software Configuration

- MLB LED Scoreboard Version: 5.1.4
- Commands/arguments used to start the scoreboard:
#412 Just hangs and nothing on the matrix

pjockey, 2022-11-05:
Trying to run a new system and getting the same issue. Difference on this unit is it is running bulseye.

WardBrian, 2023-04-10:
Is this still ongoing? Seems fine for me locally:

```
DEBUG (16:28:09): Headlines should update!
DEBUG (16:28:09): 17 feeds to update...
DEBUG (16:28:09): Feed URLs found...
DEBUG (16:28:09): Fetching https://www.mlb.com//feeds/news/rss.xml
DEBUG (16:28:09): Fetched feed 'b'MLB News'' with 25 entries.
```

closed 2025-02-22 by WardBrian