MLB-LED-Scoreboard/mlb-led-scoreboard #416: Awkward letter spacing

Bigred200, 2022-10-13:
after the use of a single column letter or number most commonly seen in the use of "i or l" there is an extra space between the next letter. Example  "framber val dez"  had to leave before I could try to adjust anything so I figured I'd post and see if there was a quick fix I could try before I just started guessing once I get home. Thanks

WardBrian, 2022-10-13:
I think this came up before and was specific to some version of the matrix library. Smaller letters like that should be centered

Bigred200, 2022-10-13:
What should I try?

ty-porter, 2022-10-13:
Certain versions of the matrix driver library left-aligned characters. I haven't followed lately to know if it's been fixed. The solution is to checkout a known good version of the driver lib and reinstall.

https://mlb-led-scoreboard.slack.com/archives/C9BLJ9JF9/p1651458034419249?thread_ts=1651432875.074229&cid=C9BLJ9JF9

c2mfj, 2023-03-03:
> 
I installed just a few weeks ago and am experiencing this, I do not have access to slack, how can I get an invite?

ty-porter, 2023-03-03:
Click the [image] button here or at the top of the README

Just to put the info from the Slack thread here so we don't lose it:

The initial recommendation 10 months ago was revert to a previous commit of the driver library
```
# from scoreboard dir
cd submodules/matrix

sudo git checkout a56338db0f003d5236f2ce98c73a591d64a70852
```

I don't recommend that any more as that commit is over a year old at this point

However, while I was testing this, our patched 4x6 font doesn't match the matrix driver 4x6 font
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/fonts/4x6.bdf#L996-L1007
https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/master/assets/fonts/patched/4x6.bdf

I think I patched the wrong font...

ty-porter, 2023-03-03:
All this to say:

I think there's a bug in `rpi-rgb-led-matrix` that left-aligns glyphs. BDF fonts have settings per-character such as bounding boxes, glyph offsets, etc., that are being ignored:

https://github.com/hzeller/rpi-rgb-led-matrix/blob/0bbf64d3c145e65ced8121befc1041c15f38c4bc/lib/bdf-font.cc#L55-L67

However, our 4x6 font is out of sync with theirs and their font's `l` and `i` glyphs have serifs which make it wider (thus aligned correctly).

ty-porter, 2023-03-03:
Alright, final update here.

BDF fonts have bounding boxes (`BBX`). The `BBX` field uses 4 values -- `WIDTH`, `HEIGHT`, `XOFFSET`, `YOFFSET`. `rpi-rgb-led-matrix` uses the first 2 and ignores the last 2.

Unfortunately for us, our 4x6 font has several common characters with a `BBX X OFFSET`, with `i` being the most noticeable. However, `.`, `!`, and some others are also affected by this bug

Example `BBX` that causes issues:
```
BBX 1 5 1 0
```

Here the `XOFFSET` is 1, meaning the glyph needs to be moved 1 pixel to the right.

[image]

---

In terms of resolution:

I'm going to put in a bug report for `rpi-rgb-led-matrix`. Not sure if/when this would get implemented.

On our end, I'll get the correct 4x6 font in place which should resolve the issue since it has no `BBX` fields with an offset.

ty-porter, 2023-03-25:
Resolved in #428.

closed 2023-03-25 by ty-porter