MLB-LED-Scoreboard/mlb-led-scoreboard #417: Only Showing Astros/Mariners Game 2 Final while game 3 is live

Bigred200, 2022-10-15:
## This issue is a

- [ ] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.):
- Operating System:
- Matrix size:
- Display adapter (HAT/Bonnet/etc.):
  - Additional information:
- Power source(s):

## Software Configuration

- MLB LED Scoreboard Version:
- Commands/arguments used to start the scoreboard:

## Expected behavior

## Actual behavior

## Additional Information

WardBrian, 2022-10-15:
Can you post your config file?

Bigred200, 2022-10-15:
> Can you post your config file?

 "preferred": {
                "teams": ["Astros"],
                "divisions": []
        },
        "news_ticker": {
                "team_offday": false,
                "always_display": false,
                "preferred_teams": false,
                "traderumors": false,
                "mlb_news": false,
                "countdowns": false,
                "date": false,
                "date_format": "%A, %B %-d"
        },
        "standings": {
                "team_offday": false,
                "mlb_offday": false,
                "always_display": false,
                "display_no_games_live":false
        },
        "rotation": {
                "enabled": false,
                "scroll_until_finished": true,
                "only_preferred": true,
                "only_live": true,
                "rates": {
                        "live": 15.0,
                        "final": 15.0,
                        "pregame": 15.0
                },
                "while_preferred_team_live": {
                          "enabled": true,
                        "during_inning_breaks": true
                }
        },
        "weather": {
                "apikey":"" ,
                "location": "",
                "metric_units": false
        },
        "time_format": "12h",
        "end_of_day": "00:00",
        "full_team_names": true,
        "short_team_names_for_runs_hits": true,
        "pregame_weather": false,
        "scrolling_speed": 2,
        "debug": false,
        "demo_date": false
}

Bigred200, 2022-10-16:
Update - I also use the NFL scoreboard and it's showing "Sunday - game " vs today -- game" so I'm gonna take a guess and say perhaps the pi isn't fetching the appropriate date/time. I'm reasonably new to this so I don't really know how to check that or fix it but that's the only guess I have right now since it's both programs. When NFL games start if it doesn't update I guess I'll know if that's right

WardBrian, 2022-10-16:
If you run the command “date” it will print out what time it think it is

Bigred200, 2022-10-17:
the NFL one does work so its not the date issue which the date is still accurate. cant see the yankees game either

ty-porter, 2022-10-17:
You have rotation disabled and only rotate through live preferred teams:

```json
"rotation": {
  "enabled": false,
  "only_preferred": true
}
```

Bigred200, 2022-10-17:
> You have rotation disabled and only rotate through live preferred teams:
> 
> 
> 
> ```json
> 
> "rotation": {
> 
>   "enabled": false,
> 
>   "only_preferred": true
> 
> }
> 
> ```

I don't want rotation, just the Astros. Even with rotation enabled it would not display the Astros game 3 live or final. Just game 2. Now nothing at all

ty-porter, 2023-03-01:
Closing, this dealt with a particular game during the offseason. Feel free to reopen if this continues.

closed 2023-03-01 by ty-porter