MLB-LED-Scoreboard/mlb-led-scoreboard #419: Only weather is being displayed no matter if games are live.

Electromag1888, 2022-10-23:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): 4b
- Operating System: 32-bit Bullseye lite, have also tried 64bit Bullseye lite as well.
- Matrix size:64x32
- Display adapter (HAT/Bonnet/etc.): Adafruit
  - Additional information: Originally was working fine.  Turned off for about a month.  Turned back on yesterday.
- Power source(s): 7amp 5.3V

## Software Configuration

- MLB LED Scoreboard Version:5.1.5
- Commands/arguments used to start the scoreboard: --led-brightness=35 --led-slowdown-gpio=4 --led-gpio-mapping="adafruit-hat"

## Expected behavior
Should display live games

## Actual behavior
Weather is displayed

## Additional Information
 No error codes at all.

Electromag1888, 2022-10-23:
This is crazy since the board was working fine when I shut it down a month or two ago.  I have even re-installed the OS and scoreboard with the same results which implies that there is either something with the MLB site, however there are no errors displayed when everything boots up.

ty-porter, 2023-03-01:
@Electromag1888 Is this ongoing now that spring training games have started back up?

ptolbert, 2023-03-01:
I have the same issue. Still weather until I manually restarted the pi.

ty-porter, 2023-03-01:
Do you have a config that I can use to try to reproduce?

pjockey, 2023-04-11:
I'm getting the only weather today. Config is below.
```
{
  "preferred": {
    "teams": [
      "Blue Jays"
    ],
    "divisions": [
      "AL East",
      "AL Wild Card"
    ]
  },
  "news_ticker": {
    "team_offday": true,
    "always_display": false,
    "preferred_teams": true,
    "traderumors": true,
    "mlb_news": true,
    "countdowns": true,
    "date": true,
    "date_format": "%A, %B %-d"
  },
  "standings": {
    "team_offday": false,
    "mlb_offday": true,
    "always_display": false,
    "display_no_games_live": true
  },
  "rotation": {
    "enabled": true,
    "scroll_until_finished": true,
    "only_preferred": false,
    "only_live": true,
    "rates": {
      "live": 15.0,
      "final": 15.0,
      "pregame": 15.0
    },
    "while_preferred_team_live": {
      "enabled": true,
      "during_inning_breaks": true
    }
  },
  "weather": {
    "apikey": "xxxxxxxxxxxxxxxxxxxx",
    "location": "Cranbrook, CA",
    "metric_units": true
  },
  "time_format": "24h",
  "end_of_day": "00:00",
  "full_team_names": true,
  "short_team_names_for_runs_hits": true,
  "pregame_weather": true,
  "scrolling_speed": 2,
  "debug": false,
  "demo_date": false,
  "preferred_game_update_delay_in_10s_of_seconds": 0
}

```

c2mfj, 2023-04-11:
Have you updated recently? If so, run the installer again.

pjockey, 2023-04-11:
> Have you updated recently? If so, run the installer again.

Yes, I've update to 6.1.3 and just checked and all is up to date. 

I just the installer again and will have to restart the board for any changes to take effect. The restarting of the board would fix things either way, so hard to tell if running the installer again actually worked.

WardBrian, 2025-03-02:
Closing old issues, please open a new one or start a discussion if running into more configuration issues

closed 2025-03-02 by WardBrian