MLB-LED-Scoreboard/mlb-led-scoreboard #421: Worked one day, then next day errors on launch

c2mfj, 2023-01-29:
## This issue is a

- [x ] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi 3
- Operating System: 64bit lite
- Matrix size: 64x32
- Display adapter (HAT/Bonnet/etc.): bonnet
  - Additional information:
- Power source(s): 5v 4000ma

## Software Configuration

- MLB LED Scoreboard Version: latest
- Commands/arguments used to start the scoreboard:  --led-gpio-mapping="adafruit-hat" --led-rows="32" --led-cols="64" --led-brightness="70"

## Expected behavior

Led board displays MLB logo then shows weather and ticker (this worked after initially installed)
## Actual behavior

MLB logo displays and stays, CLI errors below
## Additional Information

`Exception in thread render_thread:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/mlb-led-scoreboard/main.py", line 129, in __render_main
    MainRenderer(matrix, data).render()
  File "/opt/mlb-led-scoreboard/renderers/main.py", line 35, in render
    self.__render_standings()
  File "/opt/mlb-led-scoreboard/renderers/main.py", line 68, in __render_standings
    self.__draw_standings(stick=True)
  File "/opt/mlb-led-scoreboard/renderers/main.py", line 94, in __draw_standings
    self.data.standings.current_standings(),
  File "/opt/mlb-led-scoreboard/data/standings.py", line 107, in current_standings
    return self.__standings_for(self.preferred_divisions[self.current_division_index])
  File "/opt/mlb-led-scoreboard/data/standings.py", line 104, in __standings_for
    return next(division for division in self.standings if division.name == division_name)
StopIteration #shell stays here until break command is sent
^CERROR (21:09:22): Untrapped error in main!
Traceback (most recent call last):
  File "/opt/mlb-led-scoreboard/main.py", line 141, in <module>
    main(matrix, config)
  File "/opt/mlb-led-scoreboard/main.py", line 73, in main
    __refresh_standings(render, data)
  File "/opt/mlb-led-scoreboard/main.py", line 90, in __refresh_standings
    time.sleep(30)
KeyboardInterrupt`

I first got this set up a couple days ago and had no troubles following the guide. I fired it up the next day and received these errors. I decided to start from scratch and am receiving the same errors.

WardBrian, 2023-01-29:
Can you share your config json?

c2mfj, 2023-01-29:
Sure thing!

{
	"preferred": {
		"teams": ["Braves"],
		"divisions": ["NL East", "NL Wild Card"]
	},
	"news_ticker": {
		"team_offday": true,
		"always_display": false,
		"preferred_teams": true,
		"traderumors": true,
		"mlb_news": true,
		"countdowns": true,
		"date": true,
		"date_format": "%A, %B %-d"
	},
	"standings": {
		"team_offday": false,
		"mlb_offday": true,
		"always_display": false,
		"display_no_games_live": true
	},
	"rotation": {
		"enabled": true,
		"scroll_until_finished": true,
		"only_preferred": false,
		"only_live": true,
		"rates": {
			"live": 15.0,
			"final": 15.0,
			"pregame": 15.0
		},
		"while_preferred_team_live": {
			"enabled": false,
			"during_inning_breaks": false
		}
	},
	"weather": {
		"apikey": "redacted",
		"location": "Atlanta,ga,us",
		"metric_units": false
	},
	"time_format": "12h",
	"end_of_day": "00:00",
	"full_team_names": true,
	"short_team_names_for_runs_hits": true,
	"pregame_weather": true,
	"scrolling_speed": 2,
	"debug": false,
	"demo_date": false
}

WardBrian, 2023-01-30:
> "standings": {
"team_offday": false,
"mlb_offday": true,

My guess to what is happening is that it is trying to display the standings but they're currently empty/in a bad state because the season changing over from '22 to '23. MLB does not keep their API in a good state all year round, so especially in the offseason certain requests will yield unpredictable data

WardBrian, 2023-01-30:
Until spring training starts I have my personal board set to always show the news ticker. Once ST is going on the API is usually populated for the coming season. If there are any big changes we might need to create an update to account for them, but unfortunately we only know about that when it is already happening

c2mfj, 2023-01-30:
Yep that got me going. thank you!

closed 2023-01-30 by c2mfj