MLB-LED-Scoreboard/mlb-led-scoreboard #429: WBC teams added for the Baseball Classic?

jshaddo, 2023-03-07:
## This issue is a

- [ ] Bug Report
- [X] Feature Request

I've had this running in my office for 2-3 years now.  I'm glad I ran across this repository and created...   I'm wondering if the WBC teams will be added to the teams list.  I assume they would possibly be using the same MLB At Bat API.  Only an enhancement, not a bug... just curious if it would be as easy as changing the teams list.

Thanks

WardBrian, 2023-03-08:
We can get "International Baseball" easily by requesting sportId `51` in the schedule API call, but this will include duplicates of the exhibition games (e.g., on Thursday the Cardinals are playing Nicaragua) which are also listed in the normal MLB listing (sportId `1`).

I can't figure out if there's a way to request _only_ the WBC games, it seems requesting `leagueId:160` has no effect.

There would need to be a few other changes to e.g. teams.py, and obviously there wouldn't be built-in colors for them either

WardBrian, 2023-03-08:
I have a mostly-working version of this locally. There are some oddities, e.g. Columbia is also abbreviated `COL`, so it ends up purple.

Here's the relevant code:

```python
                games = statsapi.schedule(self.date.strftime("%Y-%m-%d"))

                # WBC - Id 51 is "International Baseball"
                games += statsapi.schedule(sportId="51", date=self.date.strftime("%Y-%m-%d"))

                # sort and de-duplicate
                sorted_games = sorted(
                    games, key=lambda a: datetime.strptime(a["game_datetime"], "%Y-%m-%dT%H:%M:%SZ")
                )
                latest = None
                self.__all_games = []
                for game in sorted_games:
                    if game["game_id"] != latest:
                        self.__all_games.append(game)
                        latest = game["game_id"]
```

which you can replace this line:

https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/2f400688c788e20f1ff2ee0881fa5737cf98ece5/data/schedule.py#L46

with, if you'd like. I'm not sure if we will support this "officially" or not, since it's a limited time thing and only every 4 years

jshaddo, 2023-03-13:
I've implemented this piece of code in the schedule.py file.  Everything seems to pick up the WBC games.  I need to work through shortening some of the longer names and adding custom colors for the international teams.  Thanks again!

ty-porter, 2023-03-25:
With the WBC over, I think we can close this.

closed 2023-03-25 by ty-porter