MLB-LED-Scoreboard/mlb-led-scoreboard #433: Start the Scoreboard on Autostart does't work

PHI34Halladay, 2023-03-14:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision 3B+
- Operating System: Rasbpian
- Matrix size: 128x64
- Display adapter (HAT/Bonnet/etc.): HAT
  - Additional information: -
- Power source(s): -

## Software Configuration

- MLB LED Scoreboard Version: 6.0.0
- Commands/arguments used to start the scoreboard:

## Expected behavior
The Scoreboard should load on startup after executing all of the steps the wiki page.

## Actual behavior
Nothing happens.

## Additional Information
I went through step by step in this wiki page (https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/wiki/Creating-a-systemd-startup-script) but can't seem to even get the test command (sudo service mlb-led-scoreboard start) going.
How can you help me?
[image]

pjockey, 2023-03-14:
have you tried to check and see if everything stuck. Check and see if the .service file was actually moved to /lib/systemd/system.

PHI34Halladay, 2023-03-15:
It is moved to /lib/systemd/system.

mike2me, 2023-03-18:
post screenshot of your script

PHI34Halladay, 2023-03-20:
This is how it looks like:
[image]

pjockey, 2023-03-20:
Do you have more than one board connected together? 

This is my program start:
sudo python3 main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-cols=64 --led-rows=32 --led-rgb-sequence=BGR --led-row-addr-type=3

Try running your program start from the command line and see what error are produced. Or try mine and see what happens. I'm acutally running a 128x64 board, but I changed the above to the 64x32.

PHI34Halladay, 2023-03-20:
Thanks for the response.
I'm running 4 panels (2x2) at once.

I will try our command tomorrow.

It works when I'm going like this:

1. Command: cd mlb-led-scoreboard 
2. Command: sudo python3 main.py --[......]

But not when I'm doing this:

sudo python3 home/pi/mlb-led-scoreboard/main.py --[......]

Could this be connected with the autostart function?

ty-porter, 2023-03-20:
This doesn't have anything to do with the scoreboard itself

You're bumping into issues with `systemd` to set up the service. I would recommend triple checking that the service is copied to `lib/systemd/system` and re-enabling the scoreboard service:

```
sudo systemctl enable mlb-led-scoreboard
```

Beyond that there's limited ability for us to troubleshoot for you.

PHI34Halladay, 2023-03-21:
Just saw that the file is owned by the user "pi". Could that me the mistake?
[image]

How can I change it to "root"?

PHI34Halladay, 2023-03-21:
Newest error meesage

[image]

closed 2025-03-09 by ty-porter