MLB-LED-Scoreboard/mlb-led-scoreboard #446: Add Pitch Count

pjockey, 2023-04-02:
Feature Request

## This issue is a

- [ ] Bug Report
- [X ] Feature Request

The ability to add the pitch count to a 7 segment display

[image]

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): 3B
- Operating System: Buster
- Matrix size: 128x64
- Display adapter (HAT/Bonnet/etc.): HAT
  - Additional information:
- Power source(s):

## Software Configuration

Version 6.1.3

- MLB LED Scoreboard Version:
- Commands/arguments used to start the scoreboard: sudo python3 main.py --led-gpio-mapping=adafruit-hat-pwm --led-cols=128 --led-rows=64 --led-rgb-sequence=BGR --led-row-addr-type=3

## Additional Information
If the TV sync is working, then the Pitch Count should be fairly accurate.

ty-porter, 2023-04-02:
I don't see a path forward to supporting this officially due to it requiring another piece of hardware, documentation on where to get it, how to wire it, etc.

The code itself is fairly straightforward for tweaking for personal use:
https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/0d477569584ec3bc1f8ec68d88dfd3b1329c081f/renderers/games/game.py#L92-L115

Here you can simply "render" the count wherever you like -- in a terminal log, 7 segment display, whatever.

pjockey, 2023-04-02:
Sorry, I meant the clock. Have updated the title and OP. 
I have the Pitch Count working just great.
Thanks for the work and assistance.

WardBrian, 2023-04-02:
We only update every our data every 10 seconds so a pitch clock would need to fill in our best guess in the mean time. I’m not even sure if MLB has it in the API; it’s not visible in the At Bat app for example

WardBrian, 2023-04-02:
Yeah, it seems like this data is not available in the MLB API. If you wanted to, you could in theory recreate it from the time information that _is_ in the response, and the `isPitch` field:

```
       'startTime': '2023-04-02T17:33:08.976Z',
       'endTime': '2023-04-02T17:33:13.836Z',
       'isPitch': True
```

by assuming the pitch clock started at the end time of the previous pitch and is counting down correctly, but this would 1) not work with the delay logic and 2) not be accurate whenever there is a pickoff, someone calls time, or any of the umpire-discretion delays to the clock happen

closed 2023-04-02 by WardBrian