MLB-LED-Scoreboard/mlb-led-scoreboard #462: I created a new w64h32.json and re ran install.sh but is still splitting into two 32 by 32 screens.

ianchang1, 2023-04-20:
## This issue is a

- [ ] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.):
- Operating System:
- Matrix size:
- Display adapter (HAT/Bonnet/etc.):
  - Additional information:
- Power source(s):

## Software Configuration

- MLB LED Scoreboard Version:
- Commands/arguments used to start the scoreboard:

## Expected behavior

## Actual behavior

## Additional Information

ianchang1, 2023-04-20:
I am using a Rasberry PI 3 with an adafruit matrix bonnet. Pretty new to all of this stuff so learning along the way. I got everything up and running but when playing around with configuration can't seem to have one big screen instead of two mini ones. When I rerun isntall.sh it says it updated from w64h32.json but when I boot nothing changes

c2mfj, 2023-04-20:
What are your args used to start the scoreboard?

ianchang1, 2023-04-21:
That worked thanks!

closed 2023-04-27 by ty-porter