MLB-LED-Scoreboard/mlb-led-scoreboard #463: If emoji is present in News Article Title, then python script crashes due to a ZeroDivisionError

randolph-gravely, 2023-04-20:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): N/A (RGB Emulator)
- Operating System: Mac OS
- Matrix size: 32 x 64
- Display adapter (HAT/Bonnet/etc.): N/A
  - Additional information: N/A
- Power source(s): N/A

## Software Configuration v6.3.0

- MLB LED Scoreboard Version:
- Commands/arguments used to start the scoreboard:

## Expected behavior
Ticker should rotate properly to the next screen

## Actual behavior
If emoji is present in an MLB Article, it appears it causes the script to crash (at least in emulation mode). The following error occurs:
Emulator/graphics/__init__.py", line 19, in DrawText
    adjustment = abs(x + first_char_width) // first_char_width
ZeroDivisionError: integer division or modulo by zero

## Additional Information
Please note that this issue has ONLY been tested in emulation mode! I do not have a Raspberry Pi yet to test on a physical setup. It appears the best solution would be to automatically remove emojis from MLB News Headline Names.

WardBrian, 2023-04-20:
Do you know which team’s news feed lead to the headline appearing?

swemoney, 2023-04-20:
I'm finding a play button ▶️ emoji starting to appear for "game story" articles. Just checked the astros feed

swemoney, 2023-04-20:
I'm also seeing the play buttons on the cubs feed though. I'm testing the feed on my board now and haven't experienced a crash yet

WardBrian, 2023-04-20:
The code that crashed is only in the emulator, not the actual board (I wrote it, I’m pretty sure) so I think it would need to be a different error

swemoney, 2023-04-20:
Not seeing a crash on board. Just saw the Game Story article and it ended with a play icon and then a ? symbol (like it didn't understand the character)

ty-porter, 2023-04-20:
Can you give the emulator version? It's possible this is an issue with
RGBME and not this project.

Tyler Porter

randolph-gravely, 2023-04-21:
@ty-porter how do I find the emulator version?

randolph-gravely, 2023-04-21:
I am pretty new to programming, so I am not sure if this is the best way to fix the issue, but I did resolve it by removing emojis from headlines.

I did so by adding the following to the scrollingtext.py in the renderers folder:

`from cleantext import clean`

And then at line 28:
`        clean_text = clean(text,normalize_whitespace=False, no_emoji=True, lower=False)`

randolph-gravely, 2023-04-21:
@WardBrian it was the Braves headlines that were causing the issue. It was the same thing that @swemoney described with the Astro's: it had the play emoji ("▶️ ") present in several article headlines

ty-porter, 2023-04-21:
> @ty-porter how do I find the emulator version?

```
python3 -m pip show RGBMatrixEmulator
```

ty-porter, 2023-04-21:
I can reproduce this in emulator v0.8.3

Steps to trigger the bug:
1. Character not in font
2. Missing character is the first character in the string
3. Text origin is < 0 (off matrix to the left)

For this specific trigger that would mean the string needs an emoji as the first character of a string that's large enough to scroll. The error will happen when the scrolling goes offscreen

I will fix this upstream in RGBME, however the fix there will be to gracefully handle 0 width characters. Missing characters will still show a `(?)` icon. If that's something we want to fix as well, we would need work in this project.

ty-porter, 2023-04-22:
Resolved in #465.

PR #464 (2023-04-22): Bump RGBME to 0.8.4
Resolves #463

Emulator diff available here
https://github.com/ty-porter/RGBMatrixEmulator/commit/4e85981026deebabf548daf7e887696f7b5ef22b
closed 2023-04-22 by ty-porter