MLB-LED-Scoreboard/mlb-led-scoreboard #466: Scoreboard >= v6.3.0 causes games to get "stuck" without rotating

ty-porter, 2023-04-27:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Hardware Configuration

All, as reported in Discord

## Software Configuration

- MLB LED Scoreboard Version: v6.3.0+
- Commands/arguments used to start the scoreboard: N/A

## Expected behavior
Games rotate through live games

## Actual behavior
Games occasionally stop rotating and get "stuck" on a single game.

## Additional Information
Reverting to v6.2.0 resolves, so the change must have been introduced somewhere in v6.3.0

Anyone that needs resolution immediately can checkout a known good version:

```
git checkout 6.2.0

# if using autostart service
sudo service mlb-led-scorebard restart
```

pjockey, 2023-04-27:
```
git checkout 6.2.0

# if using autostart service
sudo service mlb-led-scorebard restart
```
I have to restart the program everyday as the final scores for the will play forever if I don't. Perhaps something has happened to my config.json file.

ty-porter, 2023-04-27:
Yes, that is the issue as we have identified it. It is not a config issue to our knowledge

WardBrian, 2023-04-29:
It wouldn't hurt to share your configs. I haven't had this issue, and I wonder if there is some difference in my config which makes it not noticeable

pjockey, 2023-04-29:
```
{
  "preferred": {
    "teams": [
      "Blue Jays"
    ],
    "divisions": [
      "AL East",
      "AL Wild Card"
    ]
  },
  "news_ticker": {
    "team_offday": true,
    "always_display": false,
    "preferred_teams": true,
    "traderumors": true,
    "mlb_news": true,
    "countdowns": true,
    "date": true,
    "date_format": "%A, %B %-d",
    "display_no_games_live": false
  },
  "standings": {
    "team_offday": true,
    "mlb_offday": true,
    "always_display": false,
    "display_no_games_live": false
  },
  "rotation": {
    "enabled": true,
    "scroll_until_finished": true,
    "only_preferred": false,
    "only_live": true,
    "rates": {
      "live": 15.0,
      "final": 15.0,
      "pregame": 15.0
    },
    "while_preferred_team_live": {
      "enabled": false,
      "during_inning_breaks": false
    }
  },
  "weather": {
    "apikey": "ef96cea0f9e9806c02181548e66671d2",
    "location": "Cranbrook, CA",
    "metric_units": true
  },
  "time_format": "24h",
  "end_of_day": "01:00",
  "full_team_names": true,
  "short_team_names_for_runs_hits": true,
  "pregame_weather": true,
  "scrolling_speed": 2,
  "debug": false,
  "demo_date": false,
  "preferred_game_update_delay_in_10s_of_seconds": 0
}
```

swemoney, 2023-04-29:
My config is almost identical aside from the preferred team and standings settings (all sent to false). Also having an issue with rotation just not happening. Last night, it was stuck on the final score to the Minnesota game who's not in my preferred teams or divisions. Didn't rotate to any other final games.

swemoney, 2023-04-29:
Just to add, I've noticed multiple times since updating to 6.3.0 when the Cubs game was live and it was not staying on the Cubs game. Restarting the scoreboard, I believe, fixes it (when the game is live), and sticks to the Cubs after that.

WardBrian, 2023-04-29:
My config is different in the following ways I suspect are important:

I have more than one “preferred” team 
I have “only_preferred” set in the rotation.

ty-porter, 2023-04-29:
```
{
  "preferred": {
    "teams": [
      "Cardinals"
    ],
    "divisions": [
      "NL Central",
      "NL Wild Card"
    ]
  },
  "news_ticker": {
    "team_offday": true,
    "always_display": false,
    "preferred_teams": true,
    "traderumors": true,
    "mlb_news": true,
    "countdowns": true,
    "date": true,
    "date_format": "%A, %B %-d",
    "display_no_games_live": false
  },
  "standings": {
    "team_offday": false,
    "mlb_offday": true,
    "always_display": false,
    "display_no_games_live": true
  },
  "rotation": {
    "enabled": true,
    "scroll_until_finished": true,
    "only_preferred": false,
    "only_live": true,
    "rates": {
      "live": 15.0,
      "final": 15.0,
      "pregame": 15.0
    },
    "while_preferred_team_live": {
      "enabled": false,
      "during_inning_breaks": true
    }
  },
  "weather": {
    "apikey": "YOUR_API_KEY_HERE",
    "location": "Chicago,il,us",
    "metric_units": false
  },
  "time_format": "12h",
  "end_of_day": "00:00",
  "full_team_names": true,
  "short_team_names_for_runs_hits": true,
  "pregame_weather": true,
  "scrolling_speed": 2,
  "debug": true,
  "demo_date": false,
  "preferred_game_update_delay_in_10s_of_seconds": 6
}
```

I have similar config for those 2 keys as @pjockey

WardBrian, 2023-04-29:
Trying it now with the same config as @ty-porter (except I changed the team to the Tigers, since they have a game that's about to end)

WardBrian, 2023-04-29:
Okay, I could recreate with 6.3.1 but not with #467, so I think that PR might fix it.

PR #467 (2023-04-29): Clean up schedule advancement logic
I think this might fix #466, but I have yet to witness the issue on my own hardware so I can't fully recreate it.

A few tweaks:

1. Completed games are no longer considered "live" (we have a separate "fresh" status for them, anyway)
2. The `current_idx` is now updated if the list of games got shorter in a refresh
3. `_game_index_for_preferred_team` no longer returns `current_idx` if it fails to find a preferred game - it returns `-1`
4. Removed a flag which was set when the 'preferred game' ended (which almost certainly didn't do the right thing during doub [... cut]
PR #469 (2023-05-01): Fix re-using prior refresh of games in certain settings

This should fix #466 (for real this time almost certainly).

`self._games` should have been `games` to avoid re-using from previous refresh
closed 2023-05-03 by ty-porter