MLB-LED-Scoreboard/mlb-led-scoreboard #473: Issue when trying to set Dbacks as preferred team

stsmith313, 2023-05-18:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): 3A+
- Operating System: Raspian
- Matrix size: 32x32 & 64x32
- Display adapter (HAT/Bonnet/etc.): Bonnet
  - Additional information:
- Power source(s):

## Software Configuration

- MLB LED Scoreboard Version:
- Commands/arguments used to start the scoreboard:

## Expected behavior

Dbacks should be shown as preferred team.
## Actual behavior

There is a keyerror that happens when trying to run the program instead. If I am using the wrong team name I apologize. I was unable to find the correct team name documented anywhere.
## Additional Information

ERROR (09:15:09): Untrapped error in main!
Traceback (most recent call last):
  File "/home/sam/mlb-led-scoreboard/main.py", line 168, in <module>
    main(matrix, config)
  File "/home/sam/mlb-led-scoreboard/main.py", line 70, in main
    data = Data(config)
  File "/home/sam/mlb-led-scoreboard/data/__init__.py", line 25, in __init__
    self.current_game: Game = self.schedule.get_preferred_game()
  File "/home/sam/mlb-led-scoreboard/data/schedule.py", line 93, in get_preferred_game
    team_index = self._game_index_for_preferred_team()
  File "/home/sam/mlb-led-scoreboard/data/schedule.py", line 131, in _game_index_for_preferred_team
    team_name = data.teams.TEAM_FULL[self.config.preferred_teams[0]]
KeyError: 'Dbacks'

WardBrian, 2023-05-18:
Try the name `D-backs`

stsmith313, 2023-05-18:
> Try the name `D-backs`

Yea I've tried every iteration you can think of and same issue. Done the same thing with Arizona as well.

stsmith313, 2023-05-18:
Looks like the error does change.

ERROR (10:05:45): Untrapped error in main!
Traceback (most recent call last):
  File "/home/sam/mlb-led-scoreboard/main.py", line 168, in <module>
    main(matrix, config)
  File "/home/sam/mlb-led-scoreboard/main.py", line 70, in main
    data = Data(config)
  File "/home/sam/mlb-led-scoreboard/data/__init__.py", line 37, in __init__
    self.headlines: Headlines = Headlines(config, self.schedule.date.year)
  File "/home/sam/mlb-led-scoreboard/data/headlines.py", line 106, in __init__
    self.__compile_feed_list()
  File "/home/sam/mlb-led-scoreboard/data/headlines.py", line 183, in __compile_feed_list
    self.feed_urls.append(self.__mlb_url_for_team(team))
  File "/home/sam/mlb-led-scoreboard/data/headlines.py", line 191, in __mlb_url_for_team
    return "{}/{}/{}".format(MLB_BASE, MLB_FEEDS[team_name], MLB_PATH)
KeyError: 'D-backs'

ty-porter, 2023-05-18:
What version of the scoreboard is this running? To output:

```sh
sudo python3 version.py
```

ty-porter, 2023-05-18:
It looks like the headline keys mismatch:

https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/a593d537c7ea84bc4bfbd09ff572f52724980131/data/headlines.py#L42

ty-porter, 2023-05-18:
@stsmith313 this should be fixed on the dev branch, if you want to try it out we can release it

```sh
git checkout dev
```

Then run it, if it works for you we can go ahead and cut the actual release

stsmith313, 2023-05-18:
@ty-porter Yes! That worked.

PR #474 (2023-05-18): Handle missing headline keys, fix Windows emulated clock format
Closes #473
closed 2023-05-19 by ty-porter