MLB-LED-Scoreboard/mlb-led-scoreboard #478: Delayed game still showed start time

ZachPL, 2023-05-25:
## This issue is a

- [X] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): pi 4
- Operating System: raspos Raspberry Pi OS
- Matrix size: 32x64
- Display adapter (HAT/Bonnet/etc.): adafruit hat

## Software Configuration

- MLB LED Scoreboard Version: 6.3.2
- Commands/arguments used to start the scoreboard: --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=4 --led-cols=64 --led-brightness=70

## Expected behavior
Expected scoreboard to say the game was delayed

## Actual behavior
Still showed start time of game even though it was 45 minutes past start

## Additional Information
Happened during 5/24/23 Yankees orioles game, happened on multiple set ups I have. I searched the code and it looks like delayed game should be handled

ZachPL, 2023-06-08:
So it appears my scoreboards are not updating for irregularities, yankees game postponed due to air quality today but as of 8:30 with a gamestart of 7:05 still shows. I just restarted the scoreboard process and it shows Postponed Air Quality, so for whatever reason it isn't looking for updates when the game state changes.

tracerrx, 2025-03-09:
Closing this issue.  If the problem persists please create a new issue

closed 2025-03-09 by tracerrx