MLB-LED-Scoreboard/mlb-led-scoreboard #479: Units of measurement not working

D-Pyro, 2023-05-27:
## This issue is a

- [ ] Bug Report

## Hardware Configuration

OS: Raspbian GNU/Linux 10 (buster) armv7l
Host: Raspberry Pi 3 Model B Rev 1.2
- Matrix size: 32x64
- Display adapter Bonnet

## Software Configuration

{
  "preferred": {
    "teams": [
      "Blue Jays"
    ],
    "divisions": [
      "AL East"
    ]
  },
  "news_ticker": {
    "always_display": false,
    "team_offday": true,
    "preferred_teams": true,
    "traderumors": true,
    "mlb_news": true,
    "countdowns": true,
    "date": true,
    "date_format": "%x",
    "display_no_games_live": false
  },
  "standings": {
    "team_offday": true,
    "mlb_offday": true,
    "always_display": false,
    "display_no_games_live": true
  },
  "rotation": {
    "enabled": true,
    "scroll_until_finished": true,
    "only_preferred": true,
    "only_live": true,
    "rates": {
      "live": 15,
      "final": 15,
      "pregame": 15
    },
    "while_preferred_team_live": {
      "enabled": false,
      "during_inning_breaks": false
    }
  },
  "weather": {
    "apikey": "supersecret",
    "location": "toronto,on,ca",
    "metric_units": true
  },
  "time_format": "12h",
  "end_of_day": "00:00",
  "full_team_names": true,
  "short_team_names_for_runs_hits": true,
  "pregame_weather": true,
  "scrolling_speed": 2,
  "debug": false,
  "demo_date": false,
  "preferred_game_update_delay_in_10s_of_seconds": 0
}

- MLB LED Scoreboard Version:
MLB LED Scoreboard v6.3.4

## Expected behavior
Metric units should display in pre-game weather

## Actual behavior
Displays default units

pjockey, 2023-05-27:
I've never seen the temperature in Celsius for theBlue Jays games, even for home games. The metric shows on the "team_offdays" when the weather just stays on the screen and scrolls the news at the bottom.

WardBrian, 2023-05-27:
I think we get the weather report as a pre-formatted string from the MLB API for game weather

closed 2025-02-22 by WardBrian