MLB-LED-Scoreboard/mlb-led-scoreboard #480: Scoreboard not starting with one team config and offday

ZachPL, 2023-06-01:
## This issue is a

- [X ] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): pi 4
- Operating System: raspos Raspberry Pi OS
- Matrix size: 32x64
- Display adapter (HAT/Bonnet/etc.): adafruit hat
  - Additional information:
- Power source(s):

## Software Configuration

Version 6.3.4

{
  "preferred": {
    "teams": [
      "Yankees"
    ],
    "divisions": [
      "AL East"
    ]
  },
  "news_ticker": {
    "team_offday": true,
    "always_display": false,
    "preferred_teams": true,
    "traderumors": true,
    "mlb_news": true,
    "countdowns": true,
    "date": true,
    "date_format": "%A, %B %-d",
    "display_no_games_live": false
  },
  "standings": {
    "team_offday": false,
    "mlb_offday": false,
    "always_display": false,
    "display_no_games_live": false
  },
  "rotation": {
    "enabled": true,
    "scroll_until_finished": true,
    "only_preferred": true,
    "only_live": true,
    "rates": {
      "live": 15.0,
      "final": 15.0,
      "pregame": 15.0
    },
    "while_preferred_team_live": {
      "enabled": false,
      "during_inning_breaks": false
    }
  },
  "weather": {
    "apikey": "",
    "location": "",
    "metric_units": false
  },
  "time_format": "12h",
  "end_of_day": "00:00",
  "full_team_names": true,
  "short_team_names_for_runs_hits": true,
  "pregame_weather": true,
  "scrolling_speed": 2,
  "debug": false,
  "demo_date": false,
  "preferred_game_update_delay_in_10s_of_seconds": 0
}

- MLB LED Scoreboard Version:
- Commands/arguments used to start the scoreboard:
- sudo python3 main.py --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=4 --led-cols=64 --led-brightness=70

## Expected behavior
no error

## Actual behavior
Screen loops MLB logo shows this error

ERROR (10:59:39): Untrapped error in main!
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 168, in <module>
    main(matrix, config)
  File "/home/pi/mlb-led-scoreboard/main.py", line 70, in main
    data = Data(config)
  File "/home/pi/mlb-led-scoreboard/data/__init__.py", line 23, in __init__
    self.schedule: Schedule = Schedule(config)
  File "/home/pi/mlb-led-scoreboard/data/schedule.py", line 25, in __init__
    self.update(True)
  File "/home/pi/mlb-led-scoreboard/data/schedule.py", line 61, in update
    self.current_idx %= len(games)
ZeroDivisionError: integer division or modulo by zero

## Additional Information
Yankees are off today, and that's the only team I have showing so i'm guessing that's part of the reason why it's broken, but on past off days it has shown the weather and news so I'm not sure what's breaking it now. Happening on multiple machines with this config

ZachPL, 2023-06-01:
Changing "only_preferred": true to false allows the scoreboard to start up normally without error

ty-porter, 2023-06-04:
Thanks for the report, it does appear that having `only_preferred` turned on while the preferred teams don't have live games is the source of this bug.

The linked PR should address it, will get it released ASAP once reviewed.

PR #482 (2023-06-05): Release v6.3.5
Closes #480
closed 2023-06-05 by ty-porter