MLB-LED-Scoreboard/mlb-led-scoreboard #484: Turning off scrolling text

ianchang1, 2023-06-15:
I want to turn off the current pitcher and batter scrolling text since it is covering my team names. Where would I go to figure this out?

ianchang1, 2023-06-18:
I am using a 64 x 32 matrix with a slightly edited version of the 64x32 layout plus coordinates however, the problem was there before I edited it. Here are my scoreboard.json coordinates. I also attached a video at the bottom with the issue text is covering the team name, and there is a random square.

{
  "bases": {
    "1B": {
      "x": 50,
      "y": 11,
      "size": 10
    },
    "2B": {
      "x": 43,
      "y": 3,
      "size": 10
    },
    "3B": {
      "x": 36,
      "y": 11,
      "size": 10
    }
  },
  "final": {
    "inning": {
      "r": 255,
      "g": 235,
      "b": 59
    },
    "scrolling_text": {
      "x": 0,
      "y": 30,
      "width": 64
    },
    "nohit_text": {
      "x": 0,
      "y": 30
    }
  },
  "inning": {
    "break": {
      "number": {
        "x": 49,
        "y": 20
      },
      "text": {
        "x": 49,
        "y": 9
      }
    },
    "number": {
      "x": 62,
      "y": 30
    },
    "arrow": {
      "size": 5,
      "up": {
        "x_offset": -8,
        "y_offset": -5
      },
      "down": {
        "x_offset": -8,
        "y_offset": -1
      }
    }
  },
  "outs": {
    "1": {
      "x": 22,
      "y": 25,
      "size": 4,
      "nohit": {
        "x": 22,
        "y": 29,
        "size": 4
      },
      "perfect_game": {
        "x": 22,
        "y": 29,
        "size": 4
      }
    },
    "2": {
      "x": 29,
      "y": 25,
      "size": 4,
      "nohit": {
        "x": 29,
        "y": 29,
        "size": 4
      },
      "perfect_game": {
        "x": 29,
        "y": 29,
        "size": 4
      }
    },
    "3": {
      "x": 36,
      "y": 25,
      "size": 4,
      "nohit": {
        "x": 36,
        "y": 29,
        "size": 4
      },
      "perfect_game": {
        "x": 36,
        "y": 29,
        "size": 4
      }
    },
    "fill": {
      "1": {
        "r": 255,
        "g": 235,
        "b": 59
      }
    }
  },
  "batter_count": {
    "r": 255,
    "g": 235,
    "b": 59
  },
  "pregame": {
    "scrolling_text": {
      "x": 0,
      "y": 30,
      "width": 64,
      "warmup": {
        "x": 0,
        "y": 30,
        "width": 64
      }
    },
    "start_time": {
      "x": 49,
      "y": 15
    },
    "warmup_text": {
      "x": 49,
      "y": 15
    },
    "matchup": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "standings": {
    "divider": {
      "x": 13
    },
    "team": {
      "name": {
        "x": 1
      },
      "record": {
        "x": 31
      },
      "games_back": {
        "x": 64
      }
    },
    "nl": {
      "divider": {
        "r": 0,
        "g": 0,
        "b": 255
      }
    },
    "al": {
      "divider": {
        "r": 255,
        "g": 0,
        "b": 0
      }
    },
    "background": {
      "r": 37,
      "g": 102,
      "b": 30
    },
    "stat": {
      "r": 171,
      "g": 181,
      "b": 170
    }
  },
  "status": {
    "text": {
      "x": 49,
      "y": 15,
      "short_text": true
    },
    "scrolling_text": {
      "x": 0,
      "y": 31,
      "width": 64
    }
  },
  "offday": {
    "scrolling_text": {
      "x": 0,
      "y": 30,
      "width": 64
    },
    "time": {
      "x": 40,
      "y": 7
    },
    "conditions": {
      "x": 40,
      "y": 15
    },
    "temperature": {
      "x": 10,
      "y": 23
    },
    "wind_speed": {
      "x": 0,
      "y": -2
    },
    "wind_dir": {
      "x": 0,
      "y": -2
    },
    "wind": {
      "x": 40,
      "y": 21
    },
    "weather_icon": {
      "x": 2,
      "y": 1,
      "width": 15,
      "height": 15
    }
  },
  "network": {
    "background": {
      "width": 7,
      "height": 7,
      "x": 57,
      "y": 25
    },
    "text": {
      "x": 61,
      "y": 31
    }
  },
  "atbat": {
    "batter": {
      "r": 0,
      "g": 0,
      "b": 0
    },
    "pitcher": {
      "r": 0,
      "g": 0,
      "b": 0
    },
    "pitch": {
      "r": 0,
      "g": 0,
      "b": 0
    },
    "pitch_count": {
      "r": 255,
      "g": 255,
      "b": 255
    },
    "strikeout": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "nohit_text": {
    "r": 255,
    "g": 110,
    "b": 110
  },
  "perfect_game_text": {
    "r": 255,
    "g": 110,
    "b": 110
  },
  "default": {
    "background": {
      "r": 0,
      "g": 0,
      "b": 0
    },
    "text": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  }
}

https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/assets/131323440/e414ea7c-4ae9-493d-b8b3-425003f50171

ty-porter, 2023-06-22:
Yeah, this is a result of the custom layout. We don't support turning off arbitrary items, so to get around that you can set their coordinates off the matrix. 

You're looking for `atbat.pitcher` and `atbat.batter` coordinates options

It looks like those are missing. Have you re-run the installer? It should auto-update your configuration with all available options

```sh
sudo python sh install.sh
```

ianchang1, 2023-06-24:
That fixed it thank you!

closed 2023-06-24 by WardBrian