MLB-LED-Scoreboard/mlb-led-scoreboard #485: Option to display team record on game screen

ty-porter, 2023-07-08:
## This issue is a

- [ ] Bug Report
- [x] Feature Request

From Discord:

```
July 6, 2023 | 8:20 PM]Kevin M: hey, crew! I know I'm in the minority,
but I prefer the team initials display to the full team name.
Since there's a lot of empty space between the name and the score data,
could we add an option to display team record after it? 
BAL (50-35)  8 12 0
NYY (48-39)  0  1  0
```

ty-porter, 2023-07-14:
Released in v6.4.0

closed 2023-07-14 by ty-porter

ChickenSupreme, 2024-05-27:
I just enabled this feature on my 64x32 board and it's great! Unfortunately when both runs AND hits are >9 the record overlaps the runs and when just runs OR hits are >9 there's no empty space between them at all (and once a record includes a 3 digit number that case will overlap as well).

I decided to make a quick edit to only display the record when runs and hits are both <10. It's a pretty simple change for anyone else who wants to do the same:

Open renderers/games/team.py and look for this code (currently lines 64-65):

```
    __render_record_text(canvas, layout, away_colors, away_team, "away", default_colors, away_name_end_pos)
    __render_record_text(canvas, layout, home_colors, home_team, "home", default_colors, home_name_end_pos)
```

Change it to:

```
    if can_use_full_team_names(canvas, True, True, [home_team, away_team]):
        __render_record_text(canvas, layout, away_colors, away_team, "away", default_colors, away_name_end_pos)
        __render_record_text(canvas, layout, home_colors, home_team, "home", default_colors, home_name_end_pos)
```

I have no idea if there's enough demand for this to be added as a configurable option and this particular change is a bit of a hack anyway but I figured I'd leave this note here in case anyone else wants to apply it to their board!

As always thanks for the work you guys do on this. It's a very cool project.