MLB-LED-Scoreboard/mlb-led-scoreboard #488: Trouble starting up when plugged in

ianchang1, 2023-07-11:
Hi, I have one last problem I need fixed, thank you for all the help you have provided me so far. I am having a lot of trouble getting the system to boot on start-up despite following the exact instructions. I create the file and set the arguments, but when I do the start command nothing comes up. I tried looking into the troubleshooting already provided but I am a beginner to this type of software and I couldn't understand the errors. Please let me know what I should do to fix this issue I am happy to provide any images videos or code you might need to see of my system.

c2mfj, 2023-07-12:
When is the last time you followed the guide on the start up instructions? I recently updated them because the command syntax was outdated.

ty-porter, 2023-07-12:
Do you have specifics we can help you with?

What version of software are you running? What command are you using to run the board? Are there specific errors or stack traces you can include?

Jzbooth, 2023-07-19:
Hi, I am having a similar problem. Running the systemctl start command works perfectly if the pi is already booted and connected to the internet. However, when rebooting the pi, I get the MLB logo displaying on my LED and then the time and date with a red exclamation point in the lower right corner of the led board. It seems that it is possibly running before a stable internet connection is achieved. Any tips would be greatly appreciated.

Jzbooth, 2023-07-19:
I was able to resolve my issue using `sudo raspi-config` and selecting the option -network connection needs to be made before boot up. Hope that helps others.

ianchang1, 2023-07-22:
[IMG_2013,IMG_2011,IMG_2010.zip](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/12136930/IMG_2013.IMG_2011.IMG_2010.zip)
When checking the status errors, it says the working directory failed, so it seems like it cannot find my file in the system. I attached a photo of the errors, my file in the directory, and the code I usually boot mine up with

ty-porter, 2023-07-22:
Yes, that's the issue here.

If you can start the scoreboard manually, from the scoreboard directory run
`pwd` and update the working directory in the service with the path it
returns.

Tyler Porter

ianchang1, 2023-07-22:
That worked! Thank you for all your help

closed 2023-07-22 by ty-porter