MLB-LED-Scoreboard/mlb-led-scoreboard #497: Top Half of screen is green

DomZ31, 2023-08-10:
## This issue is a

- [X] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (3A+):
- Operating System: Raspberry Pi OS 32 Bit
- Matrix size: 64x32
- Display adapter (HAT):
  - Additional information:
- Power source(s): 2 5v 3A power converters

## Software Configuration

- MLB LED Scoreboard Version: V6.5.0
- Commands/arguments used to start the scoreboard: cd mlb-led-scoreboard; sudo python3 main.py --led-gpio-mapping="adafruit-hat" --led-cols="64" --led-pwm-bits="5" --led-brightness="30" --led-slowdown-gpio="2" --led-rgb-sequence="GBR"

## Expected behavior
 Right now it should be at the base settings; no background and going through live games and changing pitchers, hitters, and the works.

## Actual behavior
It seems to have full functionality except there is no backgrounds for teams ant the top half of board is all green writing, originally, the top had a solid green background with the teams different shades of green, now most of the time it is only showing one team in green writing. I tried changing the RGB and that would work for the bottom half as you could see the colors change with the changes of the code but the top wouldn't do it except once or twice the team turned to all blue. 

## Additional Information

[image]
[image]
[image]
[image]
[image]
[image]
[image]

pjockey, 2023-08-11:
That is a weird one. First, the error message you are getting about  isolcpus=3. Go ahead a place that statement in the /boot/cmdline.txt and reboot as the instructions state.

`sudo nano /boot/cmdline.txt`

Add isolcpus=3 and then press ctrl+x and then "y" and reboot the Pi.

As for the green, it could be the one options that you are using. I've the command below with both the 64x32 and the 128x64. Have you tried to move the letters around in this `--led-rgb-sequence="GBR"`Perhaps remove the quotation marks from the command. I don't have them in mine.

`sudo python3 main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-cols=128 --led-rows=64 **--led-rgb-sequence=BGR** --led-row-addr-type=3`

DomZ31, 2023-09-05:
I just realized I had not been refreshing this tab and didn't realize I had someone trying to help me for 4 weeks, so I am sorry. I fixed the issue you talked about and tried messing with the RGB Sequence and it definitely does not like whenever g is next to an r; rgb, brg, so on. I think there's a chance the adafruit hat or the led board spent a couple days in the mailbox in the hot texas sun and it didn't like that. Is there anywhere I can test with a multimeter for what the supposed readings should look like.

ty-porter, 2025-03-09:
Closing as this seems like it was a hardware issue, which we aren't really able to support.

closed 2025-03-09 by ty-porter