MLB-LED-Scoreboard/mlb-led-scoreboard #499: Verify software compatibility with Raspberry Pi 5

ty-porter, 2023-09-29:
https://www.raspberrypi.com/products/raspberry-pi-5/

We are looking for assistance verifying our software continues to work with the newest release of Raspberry Pi model 5.

## This issue is a

- [ ] Bug Report
- [x] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): Pi 5B revisions
- Operating System: any
- Matrix size: any
- Display adapter (HAT/Bonnet/etc.): any
  - Additional information: Please document any compatibility issues with HAT/Bonnet
- Power source(s): Raspberry Pi Foundation recommends 5V 5A power for the computer alone. The matrices typically draw quite a bit of power. Don't undervolt and brick a Pi!

## Software Configuration

- MLB LED Scoreboard Version: latest at time of RPi5 release

## Additional Information
Feel free to add compatibility issues in this thread or open new issues as needed.

pjockey, 2023-09-29:
If I can get my hands on one, I can do some testing.

ty-porter, 2025-03-09:
Tracking upstream https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

closed 2025-03-09 by ty-porter

sub-zerohm, 2025-03-11:
I got a heatsink setup for this.  Im about exhausted now, spent too much time pondering how to assemble it, was basic.  I used arctic silver instead of the little pads they give you

[image]

[image]