MLB-LED-Scoreboard/mlb-led-scoreboard #505: Error when installing

Meowscarado, 2023-10-15:
## This issue is a

- [x ] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): 3B+
- Operating System: Raspberry Pi OS Lite (32-bit)
- Matrix size: 64x32
- Display adapter (HAT/Bonnet/etc.): Adafruit RGB Matrix HAT
  - Additional information:
- Power source(s): 5V 4A

## Software Configuration

- MLB LED Scoreboard Version: 7.0.0
- Commands/arguments used to start the scoreboard: sudo ./install.sh

## Expected behavior
MLB Should install

## Actual behavior
Receive error when downloading dependencies. 

Used: sudo mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED.old

Dependencies started downloading, however, install does not install pygame.

Error:
"
Collecting pygame<3,>=2.0.1
  Using cached pygame-2.5.2.tar.gz (15.8 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [34 lines of output]
      Skipping Cython compilation

      WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
      Using UNIX configuration...

      /bin/sh: 1: sdl2-config: not found
      /bin/sh: 1: sdl2-config: not found
      /bin/sh: 1: sdl2-config: not found
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-fl76rpxc/pygame_1d9bd0136ac843d8accf14955dfe9185/setup.py", line 409, in <module>
          buildconfig.config.main(AUTO_CONFIG)
        File "/tmp/pip-install-fl76rpxc/pygame_1d9bd0136ac843d8accf14955dfe9185/buildconfig/config.py", line 231, in main
          deps = CFG.main(**kwds, auto_config=auto)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-install-fl76rpxc/pygame_1d9bd0136ac843d8accf14955dfe9185/buildconfig/config_unix.py", line 218, in main
          DependencyProg('SDL', 'SDL_CONFIG', 'sdl2-config', '2.0', ['sdl']),
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-install-fl76rpxc/pygame_1d9bd0136ac843d8accf14955dfe9185/buildconfig/config_unix.py", line 43, in __init__
          self.ver = config[0].strip()
                     ~~~~~~^^^
      IndexError: list index out of range

      Hunting dependencies...

      ---
      For help with compilation see:
          https://www.pygame.org/wiki/Compilation
      To contribute to pygame development see:
          https://www.pygame.org/contribute.html
      ---

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
"

When running sudo python3 main.py, I then get "ModuleNotFoundError: No module named 'statsapi'."

However, after manually trying to install MLB statsapi, it says it is already installed.

## Additional Information

ty-porter, 2023-10-15:
`pygame` is only installed for software emulation purposes through https://github.com/ty-porter/RGBMatrixEmulator.  You could try removing `RGBMatrixEmulator>=0.8.4` from `requirements.txt` to get it to install as a bandaid.

When you tried to install MLB StatsAPI manually, did you run as root?

```sh
sudo python3 -m pip install "MLB_StatsAPI>=1.6.1"
```

Meowscarado, 2023-10-15:
Yes I did install as root and after deleting the RBG matrix and running I have a different problem :/ 

It says my soundboard is enabled despite setting the sound to off in the config file (and rebooting), and it is saying it will not load images because the PIL is outdated... except when I upgrade pillow says I am running 10.1.0.

"
**pi@raspberrypi:~/mlb-led-scoreboard $ python -m pip install --upgrade pillow**
```
Defaulting to user installation because normal site-packages is not writeable
DEPRECATION: Loading egg at /usr/local/lib/python3.11/dist-packages/rgbmatrix-0.0.1-py3.11-linux-armv7l.egg is deprecated. pip 24.3 will enforce this behaviour change. A possible replacement is to use pip for package installation.. Discussion can be found at https://github.com/pypa/pip/issues/12330
Requirement already satisfied: pillow in /home/pi/.local/lib/python3.11/site-packages (10.1.0)
```

**pi@raspberrypi:~/mlb-led-scoreboard $ sudo python3 main.py**
```
WARNING (17:57:54): Attempted to load an insecure PIL version (9.4.0). We require PIL 10.0.1 or higher.
WARNING (17:57:54): PIL failed to load -- images will not be displayed.
Suggestion: to slightly improve display update, add
        isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)

=== snd_bcm2835: found that the Pi sound module is loaded. ===
Don't use the built-in sound of the Pi together with this lib; it is known to be
incompatible and cause trouble and hangs (you can still use external USB sound adapters).

See Troubleshooting section in README how to disable the sound module.
You can also run with --led-no-hardware-pulse to avoid the incompatibility,
but you will have more flicker.
Exiting; fix the above first or use --led-no-hardware-pulse"
```

Meowscarado, 2023-10-15:
Update, after a lot of shots in the dark regarding updates to pillow, I got it to work. Still needs tweaking since it is duplicating the screen but at least it displays something

closed 2024-03-08 by ty-porter