MLB-LED-Scoreboard/mlb-led-scoreboard #510: Just set up the hardware and I have an issue starting the software

gotopes, 2024-01-23:
## This issue is a

help request

## Hardware Configuration

- Raspberry Pi Revision 4B
- Operating System: Raspbian Debian GNU/Linux 12 (bookworm)
- Matrix size: 64x32
- Display adapter (HAT/Bonnet/etc.): Adafruit bonnet
  - Additional information:
- Power source(s): 5V 4A for matrix, 5V 3A for pi 4B
- 
- MLB LED Scoreboard Version:
MLB LED Scoreboard v7.0.0

- Commands/arguments used to start the scoreboard:

sudo python3 main.py --led-gpio-mapping="adafruit-hat"

This is what resulted:

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 3, in <module>
    from data.screens import ScreenType
  File "/home/pi/mlb-led-scoreboard/data/__init__.py", line 5, in <module>
    from data import status
  File "/home/pi/mlb-led-scoreboard/data/status.py", line 1, in <module>
    from data.scoreboard.inning import Inning
  File "/home/pi/mlb-led-scoreboard/data/scoreboard/__init__.py", line 1, in <module>
    from data.game import Game
  File "/home/pi/mlb-led-scoreboard/data/game.py", line 5, in <module>
    import statsapi
ModuleNotFoundError: No module named 'statsapi'

=============

I am not a Pi expert (just a follow the instructions person) but I sense not all the needed modules were installed.  install.sh took about 5 mins when the instructions said it should be 10-15 mins.  I figured it might be because the 4B is faster than other Pi's.

Thanks for any help.

WardBrian, 2024-01-23:
Have you tried re-running the install.sh script? It should be safe to repeat it

gotopes, 2024-01-24:
ok, i tried reinstalling.  here's all the stuff i saw on my SSH terminal.  i'd say install.sh only took about 2 mins and trying to run after the install gave the same problem I had last night.

my Python version is 3.11.2

 $ sudo ./install.sh

------------------------------------
  Installing python 3...
------------------------------------

Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-pil' instead of 'python3-pillow'
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
python3-dev is already the newest version (3.11.2-1+b1).
python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
python3-pil is already the newest version (9.4.0-1.1+b1).
python3-tk is already the newest version (3.11.2-3).
libxml2-dev is already the newest version (2.9.14+dfsg-1.3~deb12u1).
libxslt1-dev is already the newest version (1.1.35-1).
libsdl2-mixer-2.0-0 is already the newest version (2.6.2+dfsg-2).
libsdl2-image-2.0-0 is already the newest version (2.6.3+dfsg-1).
libsdl2-2.0-0 is already the newest version (2.26.5+dfsg-1).
libsdl2-ttf-2.0-0 is already the newest version (2.20.1+dfsg-2).
libopenjp2-7 is already the newest version (2.5.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

------------------------------------
  Installing dependencies...
------------------------------------

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Running rgbmatrix installation...
mkdir: cannot create directory ‘submodules’: File exists
fatal: destination path 'matrix' already exists and is not an empty directory.
Already up to date.
make -C ./lib
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[2]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
[... 35 lines cut ...]
  This operation is automatic and will ensure you have up-to-date configuration.

  This action will NOT override any custom configuration you already have unless
  the option has been obsoleted and is no longer in use.
===================================================================================

Would you like to do this now? [Y/n]
Installation finished!

$ python3 version.py
MLB LED Scoreboard v7.0.0

 $ sudo python3 main.py --led-gpio-mapping="adafruit-hat"
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 3, in <module>
    from data.screens import ScreenType
  File "/home/pi/mlb-led-scoreboard/data/__init__.py", line 5, in <module>
    from data import status
  File "/home/pi/mlb-led-scoreboard/data/status.py", line 1, in <module>
    from data.scoreboard.inning import Inning
  File "/home/pi/mlb-led-scoreboard/data/scoreboard/__init__.py", line 1, in <module>
    from data.game import Game
  File "/home/pi/mlb-led-scoreboard/data/game.py", line 5, in <module>
    import statsapi
ModuleNotFoundError: No module named 'statsapi'

WardBrian, 2024-01-24:
Oh yeah, the “externally managed” error. This is new in recent OS versions. I think some people have used that --break-system-packages flag (despite its scary name, it won’t break things in our experience) but we’ve also been meaning to update the installer so it does use a virtual environment

gotopes, 2024-01-25:
ok, where do i add the break-system-packages flag?  install.sh command from the install instructions gave an error that it was not a legit command-line parameter.

do i need to edit install.sh?  i don't understand the file entirely, but i see a section commented "installing dependancies" and some lines that have the word 'install' so do i need to put the --break-system-packages flag in each of those 3 lines using the nano editor?

WardBrian, 2024-01-25:
If you'd like to go that route, you can run

`sudo python3 -m pip install -r requirements.txt --break-system-packages`

No need to exit install.sh. We will either edit it ourselves or come up with some alternative solution, hopefully before spring training starts!

gotopes, 2024-01-26:
thanks!  it is looking much better, imho, and after running your command it seemed like modules were downloading/installing, ended up with two yellow color warnings that seemed informative, and then ran the launch command and got a weather API key message as i'd expect for not having grabbed the key yet.   now that the software seems to be launching fine with no fatal errors, i plan on re-hooking up my 64x32 matrix and check for display and add more display-related flags to the launching command.

here is what i saw on my SSH connection terminal:

 $ sudo python3 -m pip install -r requirements.txt --break-system-packages

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting feedparser==6.0.10
  Downloading https://www.piwheels.org/simple/feedparser/feedparser-6.0.10-py3-none-any.whl (81 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 81.1/81.1 kB 351.8 kB/s eta 0:00:00
Collecting MLB_StatsAPI>=1.6.1
  Downloading https://www.piwheels.org/simple/mlb-statsapi/MLB_StatsAPI-1.7.1-py3-none-any.whl (30 kB)
Collecting pyowm==3.3.0
  Downloading pyowm-3.3.0-py3-none-any.whl (4.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.5/4.5 MB 832.5 kB/s eta 0:00:00
Collecting RGBMatrixEmulator>=0.8.4
  Downloading https://www.piwheels.org/simple/rgbmatrixemulator/RGBMatrixEmulator-0.10.0-py3-none-any.whl (51 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 51.3/51.3 kB 240.3 kB/s eta 0:00:00
Collecting tzlocal==4.2
  Downloading https://www.piwheels.org/simple/tzlocal/tzlocal-4.2-py3-none-any.whl (19 kB)
Collecting sgmllib3k
  Downloading https://www.piwheels.org/simple/sgmllib3k/sgmllib3k-1.0.0-py3-none-any.whl (7.1 kB)
Requirement already satisfied: requests<3,>=2.20.0 in /usr/lib/python3/dist-packages (from pyowm==3.3.0->-r requirements.txt (line 15)) (2.28.1)
Collecting geojson<3,>=2.3.0
  Downloading https://www.piwheels.org/simple/geojson/geojson-2.5.0-py2.py3-none-any.whl (14 kB)
Collecting PySocks<2,>=1.7.1
  Downloading https://www.piwheels.org/simple/pysocks/PySocks-1.7.1-py3-none-any.whl (16 kB)
Collecting pytz-deprecation-shim
  Downloading https://www.piwheels.org/simple/pytz-deprecation-shim/pytz_deprecation_shim-0.1.0.post0-py2.py3-none-any.whl (15 kB)
Collecting bdfparser<=2.2.0
  Downloading https://www.piwheels.org/simple/bdfparser/bdfparser-2.2.0-py3-none-any.whl (12 kB)
Requirement already satisfied: pygame<3,>=2.0.1 in /usr/lib/python3/dist-packages (from RGBMatrixEmulator>=0.8.4->-r requirements.txt (line 16)) (2.1.2)
Collecting tornado>=6.1
  Downloading tornado-6.4-cp38-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (435 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 435.9/435.9 kB 845.2 kB/s eta 0:00:00

WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/libsixel-python/

Collecting libsixel-python>=0.5.0
  Downloading https://www.piwheels.org/simple/libsixel-python/libsixel_python-0.5.0-py3-none-any.whl (14 kB)
Collecting Pillow>=10.0.1
  Downloading pillow-10.2.0-cp311-cp311-manylinux_2_28_aarch64.whl (4.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.3/4.3 MB 842.4 kB/s eta 0:00:00
Collecting tzdata
  Downloading https://www.piwheels.org/simple/tzdata/tzdata-2023.4-py2.py3-none-any.whl (346 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 346.6/346.6 kB 140.0 kB/s eta 0:00:00
Installing collected packages: sgmllib3k, libsixel-python, geojson, tzdata, tornado, PySocks, Pillow, MLB_StatsAPI, feedparser, bdfparser, RGBMatrixEmulator, pytz-deprecation-shim, tzlocal, pyowm
  Attempting uninstall: Pillow
    Found existing installation: Pillow 9.4.0
    Not uninstalling pillow at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'Pillow'. No files were found to uninstall.
Successfully installed MLB_StatsAPI-1.7.1 Pillow-10.2.0 PySocks-1.7.1 RGBMatrixEmulator-0.10.0 bdfparser-2.2.0 feedparser-6.0.10 geojson-2.5.0 libsixel-python-0.5.0 pyowm-3.3.0 pytz-deprecation-shim-0.1.0.post0 sgmllib3k-1.0.0 tornado-6.4 tzdata-2023.4 tzlocal-4.2

WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

--------------

$ sudo python3 main.py --led-gpio-mapping="adafruit-hat"

Suggestion: to slightly improve display update, add
        isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
WARNING (19:20:04): [WEATHER] The API key provided doesn't appear to be valid. Please check your config.json.
WARNING (19:20:04): [WEATHER] You can get a free API key by visiting https://home.openweathermap.org/users/sign_up

closed 2024-01-27 by gotopes

k9rap, 2024-07-26:
I am having the same issue.

My scoreboard has been sitting in my closet for about two years now.  I plugged it back in and it was working just fine.

I decided to git pull to update... everything broke.

I then started with a fresh install of Raspberry Pi and followed the instructions, and the suggestion in this issue, but the app is not starting correctly.

This is the output from my ./main.py command

```
pi@scoreboard:~/mlb-led-scoreboard $ sudo ./main.py --led-gpio-mapping="adafruit-hat"
[2024-07-26 17:05:28] [RGBME] [ERROR]: 
Failed to load SixelAdapter for "sixel" display adapter!

If this is not your configured display adapter, the emulator will continue to load.

You can suppress this error in the `emulator_config.json` by adding:

  "suppress_adapter_load_errors": true

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/driver/__init__.py", line 32, in set_mode
    import rgbmatrix
ModuleNotFoundError: No module named 'rgbmatrix'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/RGBMatrixEmulator/adapters/__init__.py", line 51, in <module>
    package = importlib.import_module(package_path)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/RGBMatrixEmulator/adapters/sixel_adapter.py", line 5, in <module>
    import libsixel as sixel
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/libsixel/__init__.py", line 352, in <module>
    raise ImportError("libsixel not found.")
ImportError: libsixel not found.
[2024-07-26 17:05:32] [RGBME] [INFO]: RGBME v0.11.6 - 32x32 Matrix | 1x1 Chain | 16px per LED (SQUARE) | BrowserAdapter
[2024-07-26 17:05:32] [RGBME] [INFO]: Starting server...
[2024-07-26 17:05:32] [RGBME] [INFO]: Server started and ready to accept requests on http://localhost:8888/

```

pjockey, 2024-07-27:
Try this `sudo python3 ./main.py --led-gpio-mapping=adafruit-hat-pwm.........`

You're missing the python3 from your line.

k9rap, 2024-07-29:
hey thanks pjockey.  I did try that as one of my methods but no joy.

here is the output from that:
`pi@scoreboard:~/mlb-led-scoreboard $ sudo python3 ./main.py --led-gpio-mapping=adafruit-hat
[2024-07-29 08:27:15] [RGBME] [ERROR]: 
Failed to load SixelAdapter for "sixel" display adapter!

If this is not your configured display adapter, the emulator will continue to load.

You can suppress this error in the `emulator_config.json` by adding:

  "suppress_adapter_load_errors": true

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/driver/__init__.py", line 32, in set_mode
    import rgbmatrix
ModuleNotFoundError: No module named 'rgbmatrix'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/RGBMatrixEmulator/adapters/__init__.py", line 51, in <module>
    package = importlib.import_module(package_path)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/local/lib/python3.11/dist-packages/RGBMatrixEmulator/adapters/sixel_adapter.py", line 5, in <module>
    import libsixel as sixel
  File "/usr/local/lib/python3.11/dist-packages/libsixel/__init__.py", line 352, in <module>
    raise ImportError("libsixel not found.")
ImportError: libsixel not found.
[2024-07-29 08:27:20] [RGBME] [INFO]: RGBME v0.11.6 - 32x32 Matrix | 1x1 Chain | 16px per LED (SQUARE) | BrowserAdapter
[2024-07-29 08:27:20] [RGBME] [INFO]: Starting server...
[2024-07-29 08:27:20] [RGBME] [INFO]: Server started and ready to accept requests on http://localhost:8888/
`

k9rap, 2024-07-31:
@pjockey 
should I open a new issue for this or do you have any other suggestions?

I've wiped my raspberry pi a few times and tried anew a few times with the same issue.

I have a rasp pi 3 for what it's worth.

pjockey, 2024-07-31:
Are you running Bookworm? And followed the set up instructions on the mlb page?  I believe the install.sh creates the venv environment to house the program.
This is what all is in my mlb-led-scoreboard directory:
```
pi@scoreboard:~/mlb-led-scoreboard $ ls
assets               include                   renderers
bin                  install.sh                requirements.txt
colors               ISSUE_TEMPLATE.md         share
config.json          lib                       submodules
config.json.example  LICENSE.md                tests
CONTRIBUTING.md      main.py                   utils.py
coordinates          PULL_REQUEST_TEMPLATE.md  validate_config.py
data                 __pycache__               venv
debug.py             pyproject.toml            version.py
driver               pyvenv.cfg
highlight.py         README.md
```
The venv directory is where everything happens. My version is 8.0.2.

If you are stil having problems, I would rerun the install.sh file. Looks like the RGBMatrix is not being recognized or maybe not completely installing.

k9rap, 2024-08-02:
@pjockey, holy smokes I got it working.

I stepped through the install file section by section manually and I figured out the `make` commands from lines 138 and 139 were failing b/c of some issues with the rpi-rgb-led-matrix code.

I checked out an older version of https://github.com/hzeller/rpi-rgb-led-matrix  (sha: 2986ce6b2fa2865bf97d0083fc8a996a105eb4ff) and ran the make commands based not hat code - NOT the current master branch.

ran make commands manually and finished running the install script manually and it works.

pjockey, 2024-08-02:
Good hear. I see other posts with problems of installing the rgb matrix ans the mlb stats api. Wonder if there have been some changes in the rgb matrix?

k9rap, 2024-08-02:
> Good hear. I see other posts with problems of installing the rgb matrix ans the mlb stats api. Wonder if there have been some changes in the rgb matrix?

It stands to reason that there was a change doesn't it?

The commit I chose was kinda random. I chose one from 6 months ago, but if I had time, I would have walked it back by each commit to see which one it was.  

I do appreciate your replies and willingness to help!

Pohlmeyr, 2024-08-31:
Glad someone was able to get this working. Im very new to python. Can you help walk me through the steps that you were able to get this working? I have a PI4 with the Adafruit bonnet and Matrix 64x32. I have the same errors as you have above
Many thanks on any help you can provide