MLB-LED-Scoreboard/mlb-led-scoreboard #511: no module stats API

skhull2, 2024-01-25:
## This issue is a

- [X] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): Raspberry Pi 4 Model B - 2 GB RAM PID: 4292
- Operating System: Raspberry Pi OS (64-bit)
- Matrix size:64x32
- Display adapter (HAT/Bonnet/etc.): Adafruit RGB Matrix Bonnet for Raspberry Pi PID: 3211
  - Additional information:
- Power source(s):5V 4A (4000mA) switching power supply - UL Listed PID: 1466

## Software Configuration

- MLB LED Scoreboard Version: 7.0
- Commands/arguments used to start the scoreboard:
- When installing, run the following command in the order below:
sudo apt-get update
sudo apt-get install git python3-pip
git clone https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard.git
cd mlb-led-scoreboard/
sudo ./install.sh
sudo python3 main.py

## Expected behavior

## Actual behavior

getting error ModuleNotFoundError: No module named "statsapi" (screenshot provided)
[image]

## Additional Information

skhull2, 2024-01-26:
I ran the command sudo python3 -m pip install -r requirements.txt --break-system-packages

I then installed it again: sudo ./install.sh

I looked and it said it was externally managed (screenshot)

[image]

when I tried to define the 64x32 matrix using sudo python main.py --led-cols=64, I got the error (screenshot)
[image]

WardBrian, 2024-01-26:
That is a new error which indicates to me that you have now installed the dependencies

gotopes, 2024-01-27:
the sound module error fix is found here in the "Bad Interaction with Sound" section at https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#troubleshooting. 

On my 4B, i also had to do the blacklist steps and not just edit a config file and reboot.

alexanderthebadatcoding, 2024-02-29:
I had a similar issue, I tried to install it manually and it says:
```
pip3 install statsapi
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting statsapi
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/statsapi/
```

alexanderthebadatcoding, 2024-02-29:
I had a similar issue I did this 

` sudo python3 -m pip install --upgrade pip
`

and that fixed it

ty-porter, 2024-03-08:
Closing as this seems like it's a known issue (#510) and been addressed manually.